dotnet-bluetooth-le/dotnet-bluetooth-le #11: getting stared with plugin

Gydh1se, 2016-01-20:
Hi 

I am trying to make use of the MvvmCross.Plugins.BLE plugin in a Xamarin forms project.  I am having trouble getting stared.  Especially in regard to the injecting of the IAdapter service in my project.  Would appreciate help getting stared.  Perhaps even a sample project would be greatly appreciated.

Thanking you in advance

xabre, 2016-01-20:
Hi

Did you setup MvvmCross correctly to work with xamarin forms? How do your setup files look like?

If you are not using MvvmCross with your project it will be a little more complicated. You will have to create the MvvmCross setup files and intialize the IoC container. 

A pure Xamarin cross platform plugin is planned.

Gydh1se, 2016-01-20:
Hi, thanks for the speedy reply

I am fairly new to Xamarin and not familiar with MvvmCross.  In my Xamarin.forms project I install the MvvmCross.Plugin.BLE plugin, from the the NuGet gallery.  As a result, of the package installation, the following packages were also installed:
MvvmCross.HotTuna.CrossCore
MvvmCross.Plugin.BLE
MvvmCross.PortableSupport
Newtonsoft.Json 
 
See link (https://github.com/Gydh1se/bluetooth_test) to find a project that I created to illustrate my setup.  The project is basically just a blank forms project where I installed the MvvmCross.Plugin.BLE plugin.

Could you possibly help me from here? Not sure how I would go about letting the MvvmCross inject the IAdapter service in to the shared code.

Thanking you in advance

xabre, 2016-01-20:
Ok I understand. 
In that case here's what you have to do:

### Get latest prerelease version of the BLE plugin ###
* Remove current packages
* Install the prerelease version 0.9.0-beta3, it will install also part of mvvmcross 4.0.0-beta8

### Basic setup of MvvmCross ###
* Install MvvmCross Core 4.0.0-beta8
* Install MvvmCross Bindings 4.0.0-beta8
* Add a simple setup file
``` 
public class Setup : MvxAndroidSetup
    {
        public Setup(Context applicationContext)
            : base(applicationContext)
        {

        }

        protected override IMvxApplication CreateApp()
        {
            return new MvxSimpleEmptyAndroidApp();
        }
    }
```

* Initialize the Setup in MainActivity.cs

```
protected override void OnCreate (Bundle bundle)
		{
			//Xamarin.Insights.Initialize (XamarinInsights.ApiKey, this);
			base.OnCreate (bundle);

            var setup = new Setup(ApplicationContext);
            setup.Initialize();

			global::Xamarin.Forms.Forms.Init (this, bundle);
			LoadApplication (new App ());
		}
```

### Resolve the adapter ###
* Use IoC container provided by MvvmCross to resolve the implementation of the IAdapter interface in the portable class library.
* Modify your ble_test.cs to include this:
```
public App ()
		{
			
			// The root page of your application
			MainPage = new ContentPage {
				Content = new StackLayout {
					VerticalOptions = LayoutOptions.Center,
					Children = {
						new Label {
							XAlign = TextAlignment.Center,
							Text = "Welcome to Xamarin Forms!"
						},
                        new Button()
                        {
                            Text = "Do Ble Stuff",
                            Command = new Command(DoBleStuff)
                        }
					}
				}
			};
		}

	    public void DoBleStuff()
	    {
            //Resolve adapter
	        var adapter = Mvx.Resolve<IAdapter>();
            
            ///Do stuff
	    }
```

Note: MvvmCross is a powerfull set of tools for building cross platform apps. One of it's features is a IoC container which we used here to manage the plugin.

In the future there will be a non-MvvmCross version of this plugin.

xabre, 2016-05-13:
Check the new sample app and documentation in v 1.0

closed 2016-05-13 by xabre