dotnet-bluetooth-le/dotnet-bluetooth-le #13: Xamarin Plugin

xabre, 2016-02-05:
I would also like to create a pure Xamarin plugin with no dependencies on MvvmCross. 

My initial idea is to either use precompiler directives like #if #endif to isolate MvvmCross dependencies and create 3 new csprojects which link the same files as the existing ones.

What are your thoughts on this? :)

adolgov, 2016-02-05:
Could split it up such that there are the core "pure Xamarin" projects, and then the MvvmCross ones that, essentially, wrap the MvvmCross Plugin stuff around the base Xamarin code. i.e. when selecting the MvvmCross NuGet package, it'd have a dependency on the pure Xamarin NuGet package. The added MvvmCross wrapper package would give you the benefit of MvvmCross IoC injection stuff and Plugin registration, etc., if you were already in an MvvmCross ecosystem.

One thing that I don't know is how many Mvx things are in the code. I know that there are plenty Mvx.Trace type calls, but those are potentially easy to de-couple by having a "base" implementation (as annoying as that'd be) in the "pure Xamarin" project (probably just going to Console.Write or something like that), which would be overriden to the existing full Mvx implementation in the MvvmCross project.
If the existing code has more advanced Mvx calls (e.g. Mvx.Resolve or something like that), that could, certainly, be more tricky to get around without rewriting code.

smstuebe, 2016-04-08:
Split up into #19 #20 #18

closed 2016-04-28 by xabre