dotnet-bluetooth-le/dotnet-bluetooth-le #14: Windows Support

georgeedwards, 2016-02-10:
Are there any plans to expand the scope of this project to include the Windows platforms (8.1, phone 8.1 and/or Universal Windows)?

smstuebe, 2016-02-11:
Hey I bought a BLE Dongle for Windows last week. I plan to add UWP support :) But pssst. Don't tell @xabre ;)

xabre, 2016-02-11:
Very nice. Just wanted to answer that unfortunately I have no test device.
Sven, can you share a link to the dongle you ordered?

smstuebe, 2016-02-11:
Yes, after I did the first tests. I don't want to recommend it if its shit ^^

xabre, 2016-02-11:
Fair enough. :) I will also open an issue regarding a sample app. Maybe for
all 3 platforms :)

georgeedwards, 2016-02-11:
Great, well for my project, that would be really useful, I am not familiar with this plugin, but will be looking over it through the next few days, let me know if there is anything I can do to help?

smstuebe, 2016-02-15:
work is going on in uwp branch :smile_cat: in combination with #15 :smirk_cat: and some refactorings :scream_cat:

georgeedwards, 2016-04-01:
Sorry, I am back now (last few weeks have been super busy). @xabre Have you had any luck with the sample app? I am playing around with the existing project now to get my head around it and this would be really handy, if not - no worries.

soernt, 2016-05-27:
Hi there,
last year I did the same. I take the Monkey robotics, refactored it and worte impl. for iOS, Android and Windows Phone (WinRT). Maybe I can help here.

xabre, 2016-05-27:
@georgeedwards The sample app is up and running. You can find it in the solution ([Client](https://github.com/xabre/xamarin-bluetooth-le/tree/master/Source/BLE.Client))

DumDumin, 2016-07-18:
Hey guys,
did you quit on uwp support?

smstuebe, 2016-07-20:
We freezed it, because we wan't a neat api for ios and android first. And UWP will be a bit pain in the ass, because it is following a complete different approach when it comes to connection handling.

DumDumin, 2016-07-21:
oh thats a pitty..
I am having trouble finding crossplatform library which supports all three platforms. I guess, I will end up coding the uwp implementation myself then.

Is there a simple explanation how this call is working: "var ble = CrossBluetoothLE.Current;" in the Vanilla package? How does is resolve the right "BleImplementation" at runtime?

I am still a beginner with crossplatform development and the main thing I know is the "DependencyService", which comes with Xamarin. With this you would have to call: "DependencyService.Register<IBluetoothLE, BleImplementation>" in all three plattform specific main functions (MainActivity.cs, AppDelegate.cs, App.xaml.cs). After that I can get the right implementation with: "DependencyService.Get<IBluetoothLE>();".

I hope, that these question wont give you too much trouble.
Thanks

smstuebe, 2016-08-05:
@DumDumin The vanilla Xamarin Plugins use the Bait & Switch Pattern (http://log.paulbetts.org/the-bait-and-switch-pcl-trick/) to resolve the implementation.

Grundge, 2016-08-05:
Hello @smstuebe does this mean that WP will be available with BLE? I installed it from the Nugget gallery, and got it implemented in Android and iOS, but WP fails with the the error that @DumDumin said. If it is still possible to run on WP, could you elaborate a little but more about what you just posted?

smstuebe, 2016-08-05:
No. I said: 

> We freezed it, because we wan't a neat api for ios and android first.

behradizadi, 2016-08-27:
@smstuebe  I've installed windows 10 core iot on Raspberry Pi 3, i need to use your library to connect to that, but sadly Raspberry is not shown in the bluetooth list.

I really appreciate if you could advise in this matter

sarkreth, 2016-11-16:
@smstuebe Would future development on UWP support be targeted at using a dongle or native API?
> Hey I bought a BLE Dongle for Windows last week. I plan to add UWP support :)
If this is still going to be the case when development proceeds, why?

smstuebe, 2016-11-16:
The UWP implementation will make use of the native API. The dongle is just for me to make my dev machine BLE ready ^^

The current promblems are:
- The UWP API is designed in a way that makes it hard to fit into the current structure (connection handling is weird)
- Adrian and I, currently can't spend as much time as we want on it.

flyingxu, 2017-07-03:
Hi Everyone,

I'd like to help with the UWP support.

As mentioned in #158 , the Bluetooth on UWP (Windows 10, Creators Update) is now very similar to Android/iOS. It's a good time now to write some code for UWP bluetooth.

I've tried to create a fork and a `uwp.forms` branch to see if it's possible. It turned out to be much simpler than I thought. You can have a look at my [uwp.forms branch](https://github.com/flyingxu/xamarin-bluetooth-le/tree/uwp.forms/Source/BLE.Client/BLE.Client.UWP). I'm still working on it and here is the progress:
1. UWP UI is working. Xamarin.Froms for UWP is used, so UI is almost ported completedly from Android/iOS project, I don't remember I changed any code on the UI part.
2. Bluetooth scan, connection, get services, and get charactoristics are working now (very limited, but it's working)
3. todo: implement charactoristics write/update reading.

I'd like to create a PR and put my work into this repo now or maybe later, so I have several quesitons to you all:

1. Is it a good time to create PR now? 
My branch is still in an early implementation phase, but I feel it's very promising to be completed without big design changes.
2. I use MvvmCross 5. 
It's easier for me to make Xamarin.Forms working in UWP by MvvmCross 5. I tried, but I just can't get MvvmCross 4.4 work with Xamarin.Forms in UWP. I also tried to upgrade the Android project to MvvmCross 5, it's working. I have not tested the iOS project with MvvmCross 5 yet, but I will try.

3. My branch is based on the `master` branch, not the `uwp` branch.
The `uwp` branch seems to be very out dated, there are lots of comflicts even when I tried to merge the master branch to the `uwp` branch. So I gave up and start from `master` branch.

4. I'm not very familiar with nuget things.
I think I need to change some nuget setting to make this package installable in a uwp project, but I don't know how. I may need help on that part.

During my try, I felt the API design is very good, I just followed the base classes to create the uwp specific implementation without any need to change the API. The new UWP Bluetooth API is also very good, when the API needs an async implementation, there is an async UWP API right there, I felt it's even more fluent than Android/iOS callback/event implementations.

If the coding conventions are concerned, it should not be an issue. I'm planning to to some refactoring about some of the function names and I tried to follow current style. Maybe I missed some rules, but I'm very open if some changes are needed. Now there are good tools, it's not a tedious work to do refactory work these days.

I'd like to hear your comments.
Thanks.

DumDumin, 2017-07-06:
I would love to see this library finally supporting UWP :)

DumDumin, 2017-07-17:
I tried your branch. The connecting is working pretty solid, but there is no proper way to disconnect from a device. If I close my application, the LE device still thinks its connected and cannot be found again...

https://docs.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.bluetoothledevice
this documentation includes a method "close" but thats not implemented, i dunno another way to properly disconnect a device.

xabre, 2019-08-11:
UWP is available as part of the 2.2.0-pre1 release.
ToDo: 

- [ ] Investigate disconnect/close

sarkreth, 2019-08-12:
Windows will reliably disconnect if you call the "Dispose" methods of all Services and "BluetoothLEDevice"s accessed.

Auto72, 2020-04-03:
With the version 2.2.0-pre2, in the UWP application project, the following call doesn't close the connection and never ends:

await _adapter.DisconnectDeviceAsync(_device);

As a workaround I tried to use instead:

_characteristic = null;                    
_service.Dispose();
_service = null;                    
//await _adapter.DisconnectDeviceAsync(_device);
_device.Dispose();
_device = null;

But it doesn't work.

janusw, 2023-10-09:
> With the version 2.2.0-pre2, in the UWP application project, the following call doesn't close the connection and never ends:
> 
> await _adapter.DisconnectDeviceAsync(_device);

The disconnect issues in UWP should have been fixed in 3.0.0-beta.5 and beta.6.

Windows support is available in the stable version 3.0.0 since yesterday.

closed 2023-10-09 by janusw

Auto72, 2023-10-10:
1) The await _adapter.DisconnectDeviceAsync(_device) now works and it takes at least 3 seconds to complete.

2) If my BLE device is not paired on Windows, I can't connect with it. I receive the following error message:

[image]

If I do the pairing manually on Windows, then I can connect with it.
On Android and iOS, if my BLE device is not paired, when I try to connect to it, the Plugin.BLE and OS take care of the pairing process automatically. It would be nice if this happens also with Windows. 
Here some Windows API code to pair with the BLE device:

private BluetoothLEDevice? _bleDevice;

            // Do not allow a new Pair operation to start if an existing one is in progress.
            if (!_bleDevice.DeviceInformation.Pairing.IsPaired)
            {
                if (_bleDevice.DeviceInformation.Pairing.CanPair)
                {
                    _bleDevice.DeviceInformation.Pairing.Custom.PairingRequested += CustomOnPairingRequested;
                    var result = await _bleDevice.DeviceInformation.Pairing.Custom.PairAsync(DevicePairingKinds.ConfirmOnly, DevicePairingProtectionLevel.Encryption);

                    if (result.Status != DevicePairingResultStatus.Paired)
                    {
                        msg = $"Pairing failed for Device Id: [{_bleDevice.DeviceId}] Status: [{result.Status}] ProtectionLevelUsed: [{result.ProtectionLevelUsed}]";
                        Debug.WriteLine(msg);
                    }
                    DevicePairingResult dpr = await _deviceInfo?.Pairing.PairAsync();
                    if (dpr.Status == DevicePairingResultStatus.Paired || dpr.Status == DevicePairingResultStatus.AlreadyPaired) 
                    { 
                        Debug.WriteLine($"Device paired."); 
                    }
                    else 
                    {
                        Debug.WriteLine("Unable to pair the device.");
                    }
                }
            }

private void CustomOnPairingRequested(DeviceInformationCustomPairing sender, DevicePairingRequestedEventArgs args) => args.Accept();

janusw, 2023-10-10:
> 1. The await _adapter.DisconnectDeviceAsync(_device) now works and it takes at least 3 seconds to complete.
> 
>     2. If my BLE device is not paired on Windows, I can't connect with it. I receive the following error message:

@Auto72 This issue is closed. Please open new issues for any bugs or enhancement requests ...