dotnet-bluetooth-le/dotnet-bluetooth-le #16: Namespace name 'Touch' does not exist in the namespace 'MvvmCross.Plugins.BLE

georgeedwards, 2016-04-01:
I have just installed version 0.9.1 into a Xamarin.Forms PCL project, targetting ios & android. However, when I run an ios build, I get the following error.

>Error: CS0234	The type or namespace name 'Touch' does not exist in the namespace 'MvvmCross.Plugins.BLE' (are you missing an assembly reference?)	My_App.iOS	C:\Users\George\Desktop\My_App_Xamarin_Forms\iOS\Bootstrap\BlePluginBootstrap.cs

I have a few other issues I am running into with my project, so it could be related to those, but It struck me as an odd thing to get, is this a know issue or a common problem - any idea what might be causing it?

adolgov, 2016-04-01:
Just an outdated namespace reference. Should be "iOS" instead of "Touch". I noticed that myself yesterday.

xabre, 2016-04-02:
Fixed in the 0.9.2 release.

closed 2016-04-02 by xabre