dotnet-bluetooth-le/dotnet-bluetooth-le #17: Testing Support Library

smstuebe, 2016-04-06:
In our projects we are always writing mocks for IDevice, IAdapter, ... 
We should provide a extra package where we offer some useful mocks.

xabre, 2016-04-06:
I have just the thing :)

smstuebe, 2016-04-06:
Ye, but you know.... It needs "refactoring" ^^

xabre, 2016-04-07:
Was thinking of adding this feature as a separate nuget package whith a dependecy on the PCL library.

smstuebe, 2016-04-07:
Ye that is what I meant :>

cbrammer, 2016-05-15:
What is the word on this one? I would be very keep to see how @smstuebe implemented your mocks?

smstuebe, 2016-05-15:
I think this will evolve, when we add some unittests. We want to harden the changes done with 1.0. The 1.0 refactoring happened on our "camp" so we focused on functionality.