dotnet-bluetooth-le/dotnet-bluetooth-le #34: Startupdate not working on reconnect using the extension

KevinPou, 2016-04-28:
Hey guys,
Great job on the API. I initially grabbed the original project and heavily modified to make it work on my side. When I saw your code I decided to switch as I like this API much more (has support for API 21).

However I ran into an issue that took me a couple of hours to figure out. If the connection is lost, using this code will reconnect to the device but setCharacteristicNotification will fail hence StartUpdates() on Charactrastics won't work. 

Assume DeviceId was saved. In my case it is a preferred device set by users which is meant be auto-connected to. 

var device = await Adapter.DiscoverSpecificDeviceAsync (deviceId);
var connectedDevice = await Adapter.ConnectAsync (device);
var service = await connectedDevice.GetServiceAsync (SOME_GATT_SERVICE_ID);
_sampleCharacteristic = await service.GetCharacteristicAsync (SOME_GATT_CHAR_ID)
_sampleCharacteristic .StartUpdates();

This fails since the DiscoveredDevices list is not cleared hence Adapter.DiscoverSpecificDeviceAsync returns the stored IDevice and Gatt object. 

Not sure this is intentional or not. To fix this in my code, I listen to ConnectionLost event and clear the  DiscoveredDevices list.

xabre, 2016-04-28:
Thanks for pointing that out :). We'll investigate the issue as soon as possible and release it with 1.0.

KevinPou, 2016-04-28:
Thanks! This is where I got to figure out my issue, under GattCallback.cs of the Droid project:

if (DeviceOperationRegistry.TryGetValue(gatt.Device.Address, out device))
                    {
                        Mvx.Trace("Disconnected by user");

                        //Found so we can remove it
                        DeviceOperationRegistry.Remove(gatt.Device.Address);
                        ConnectedDeviceRegistry.Remove(gatt.Device.Address);
                        gatt.Close();

                        DeviceDisconnected(this, new DeviceConnectionEventArgs { Device = device });
                        break;
                    }

                    //connection must have been lost, bacause our device was not found in the registry but was still connected
                    if (ConnectedDeviceRegistry.TryGetValue(gatt.Device.Address, out device))
                    {
                        Mvx.Trace("Disconnected by lost connection");

                        ConnectedDeviceRegistry.Remove(gatt.Device.Address);
                        gatt.Close();

                        DeviceConnectionLost(this, new DeviceConnectionEventArgs() { Device = device });
                        break;
                    }

User disconnect clears out both connected and discovered devices while lost connection only clears out the connected devices. Again, thanks for looking into it

closed 2016-05-13 by xabre