dotnet-bluetooth-le/dotnet-bluetooth-le #35: IDevice.ReadRssi() throws NullReferenceException

Cheesebaron, 2016-05-04:
I am playing around with the plugin a bit and ran into this issue with calling ReadRssi() on an IDevice instance.

What I am doing is basically, scan for devices, select one of them and show details about it.

On that selected IDevice instance I do:

    _device?.ReadRssi();

This throws the following exception:

```
05-04 10:43:20.080 E/mono-rt (20530): [ERROR] FATAL UNHANDLED EXCEPTION: System.NullReferenceException: Object reference not set to an instance of an object.
05-04 10:43:20.080 E/mono-rt (20530):   at MvvmCross.Plugins.BLE.Droid.Bluetooth.LE.Device.ReadRssi () [0x00000] in c:\Users\sead\Documents\GitHub\MvvmCross-BluetoothLE\Source\MvvmCross.Plugins.BLE.Droid\Bluetooth\LE\Device.cs:264 
05-04 10:43:20.080 E/mono-rt (20530):   at com.bksv.ems.sxuble.ViewModels.DeviceDetailsViewModel.StartData () [0x00031] in C:\Users\Tomasz\Documents\Visual Studio 2015\Projects\SXUBLE\SXUBLE\ViewModels\DeviceDetailsViewModel.cs:90 
05-04 10:43:20.080 E/mono-rt (20530):   at com.bksv.ems.sxuble.droid.Views.DeviceDetailsView.OnResume () [0x00008] in C:\Users\Tomasz\Documents\Visual Studio 2015\Projects\SXUBLE\SXUBLE.Droid\Views\DeviceDetailsView.cs:22 
05-04 10:43:20.080 E/mono-rt (20530):   at Android.App.Activity.n_OnResume (IntPtr jnienv, IntPtr native__this) [0x00009] in /Users/builder/data/lanes/2923/b4be22e9/source/monodroid/src/Mono.Android/platforms/android-23/src/generated/Android.App.Activity.cs:3920 
05-04 10:43:20.080 E/mono-rt (20530):   at (wrapper dynamic-method) System.Object:ec6c167c-6312-4270-86db-3b0ff9ed091e (intptr,intptr)
```

Looking at the line it throws at: https://github.com/xabre/xamarin-bluetooth-le/blob/master/Source/MvvmCross.Plugins.BLE.Droid/Bluetooth/LE/Device.cs#L264 it could seem like the `_gattCallback` is null for some reason.

Cheesebaron, 2016-05-04:
The same thing happens with ReadRssiAsync as it calls the same method internally.

smstuebe, 2016-05-04:
Thanks. We will check this in the 1.0 implementation. Reading RSSI changed a bit.

xabre, 2016-05-04:
This happens because of the following reason (android): 

- If a device is connected it gets an associated BluetoothGatt and GattCalback. The android ReadRemoteRssi is a method of BluetoothGatt so it works only if the device is connected.
- ReadRssi can't be called on IDevices returned by discovery and not connected (because of no gatt)... 

We have to find a way to make this clearer in the API.

Cheesebaron, 2016-05-04:
I guess XML docs would be a good place to document this. Or a small wiki walking this through.

smstuebe, 2016-05-04:
That's all on our todo :) I hate plugins without documentation

smstuebe, 2016-05-24:
@Cheesebaron is is resolved in your opinion? Does it work and feel comfortable to use in your use case, now?

Cheesebaron, 2016-05-25:
Yes, it is good now 👍

closed 2016-05-25 by smstuebe