dotnet-bluetooth-le/dotnet-bluetooth-le #36: Only works on Android Lollipop or above.

allbdeveloper, 2016-05-07:
Hi, Very Thanks for this package.

Are you planning to support for KitKat or Jelly Bean.
It's crash on those version.

smstuebe, 2016-05-07:
Where does it crash? Do you have a crash log and the code that causes the crash log? On which device are you testing?

smstuebe, 2016-06-14:
More than one month no answer -> close

allbdeveloper, 2016-06-16:
Hi, I’m so sorry that I reply so late.
I know you have put big effort to make library. 

So sorry again. 

BLE Adapter crash on Jellybean and Kitkat. (Both are tested on Xamarin Android Player - not actual device)

- JellyBean

- kitkat

Thanks.

smstuebe, 2016-06-16:
Ok. We have to review that. Have you tested it on physical devices, too?

allbdeveloper, 2016-06-16:
Hi, Thanks for reopening issue. 

Not yet. 

I will try it asap and send the result to you.

Thanks.

smstuebe, 2016-06-16:
And: You say it's crashing. Please provide the exception messages and the stacktrace.

allbdeveloper, 2016-06-16:
Hi, I attached screenshot with crash log. but it's not shown here. (I attached it using email)
DId you see that screenshot from email box? and you want more information? (exception messages & stacktrace) 
If so, I will send it.

Plus, It may take little longer to find actual device (Kitkat and Jellybean)

allbdeveloper, 2016-06-16:
<Jelly Bean>
[image]
[image]
[image]

<Kit Kat>
[image]
[image]

allbdeveloper, 2016-06-17:
I tested it on actual device. 

Samsung Galaxy Note 1. (SHV-E160L)
It crash on this device. 

[image]
[image]
[image]
[image]

allbdeveloper, 2016-06-17:
Hi, 

I tested it today on actual device (Galaxy Note 1)
Android 4.1.2 - Jelly Bean.

It crash. 

And I uploaded screenshots on github’s issue.

Thanks.

allbdeveloper, 2016-06-18:
Oh, I found it from wikipedia .
https://en.wikipedia.org/wiki/Bluetooth_low_energy#Android

BLE's supported from Android 4.3 JellyBean or above.
So It shouldn't work on 4.1 JellyBean.

xabre, 2016-06-23:
Android - Check BLE availability by:
- [ ] Checking SDK level and notify via log message
- [ ] Check if Emulator and notify via log message

smstuebe, 2016-06-23:
We should try do not check if it's a simulator/emulator. Instead we should just check if ble is available.

xabre, 2016-08-05:
You should not instantiate the Adapter by calling new. 

- If you use the vanilla version (non mvvmcross) of the plugin please resolve IAdapter by calling `CrossBluetoothLE.Current.Adapter `
- For MvvmCross use Mvx.Resolve<IAdapter>

For both cases, if** BLE it's not available on your device** you will get a fake adapter which will not crash and print out error messages.

You can always check availability by calling` CrossBluetoothLE.Current.IsAvailable`

For more info see our readme.

closed 2016-08-05 by xabre

allbdeveloper, 2016-08-27:
Thanks for your answer.