dotnet-bluetooth-le/dotnet-bluetooth-le #39: Characteristic.WriteAsync never returns on iOS

Cheesebaron, 2016-05-17:
## Steps to reproduce

1. Connect to device
2. Discover services
3. Discover characteristic
4. Write to characteristic async

My code does the following:

```
_bleAdapter.ConnectToDevice(device);
```

In the DeviceConnected event I do the following:

```
private async void BleAdapterOnDeviceConnected(object sender, DeviceConnectionEventArgs args)
{
    var device = args.Device;

    var deviceViewModel = Devices.FirstOrDefault(d => d.Id == device.ID);
    deviceViewModel.Connected = true;
    deviceViewModel.DeviceName = device.Name;

    Mvx.TaggedTrace("HomeViewModel::DeviceConnected", $"Connected to {deviceViewModel.DeviceName} - UUID {deviceViewModel.Uuid}");

    try
    {
        var service =
            await device.GetServiceAsync(new Guid("13333333333333333333333333333337")).ConfigureAwait(false);
        if (service != null)
        {
            Mvx.TaggedTrace("HomeViewModel::DeviceConnected", "Found WiFi configuration service");

            var characteristic =
                await
                    service.GetCharacteristicAsync(new Guid("13333333333333333333333333330001"))
                        .ConfigureAwait(false);

            if (characteristic != null && characteristic.CanWrite)
            {
                Mvx.TaggedTrace("HomeViewModel::DeviceConnected", "Found SSID configuration characteristic");

                var ssidBytes = Encoding.UTF8.GetBytes(HotspotViewModel.Ssid);
                var writeOk = await characteristic.WriteAsync(ssidBytes).ConfigureAwait(false);
                if (writeOk)
                {
                    Mvx.TaggedTrace("HomeViewModel::DeviceConnected", "Wrote SSID successful!");
                    deviceViewModel.Ssid = HotspotViewModel.Ssid;
                }
            }
        }

        _bleAdapter.DisconnectDevice(device);
        Mvx.TaggedTrace("HomeViewModel",
            $"Disconnecting device  {deviceViewModel.DeviceName} - UUID {deviceViewModel.Uuid}");
    }
    catch (Exception e)
    {
        Mvx.TaggedTrace("HomeViewModel::DeviceConnected", $"Got exception: {e}");
    }
}
```

The code never reaches the `if (writeOk)` and does not throw any exceptions either.

This same construction works fine on Android.

## Expected behavior
WriteAsync should eventually return

## Actual behavior
WriteAsync never returns

## Configuration

**Version:** MvvmCross.Plugin.BLE 0.9.2

**Platform:** iOS

**Device:** iPhone 4s, iPhone 6+

smstuebe, 2016-05-17:
Maybe moving the declaration of the event handler `writtenHandler` to the local space is enough

https://github.com/xabre/xamarin-bluetooth-le/blob/master/Source/Plugin.BLE.iOS/Characteristic.cs#L72

Cheesebaron, 2016-05-17:
I tried with latest 1.0.0-beta and the issue is still there.

smstuebe, 2016-05-24:
@Cheesebaron  Unfortunately, we have no test device with a WriteWithResponse characteristic. Could you please use the sample app, connect and try to write a value?

adolgov, 2016-05-24:
I have seen a similar issue on iOS. We were trying to have our peripheral device support both WriteWithResponse and WriteWithoutResponse (which is legal in BLE, from what I can tell), and iOS would report WriteWithResponse on the characteristic, but not actually ever send a response. It's possible that the outgoing write didn't identify as trying to write with response, but I didn't look into it much further.

We got around the issue by only enabling WriteWithoutResponse on our peripheral device, but that's not a solution for most people dealing with existing peripherals which the do not control.

smstuebe, 2016-05-24:
Off topic: @adolgov thanks for your recent comments. Are you at https://xamarinchat.slack.com/ or can you send me an message on twitter?

commit dc860a1c (2016-05-24): attempt to resolve #39
closed 2016-05-26 by smstuebe