dotnet-bluetooth-le/dotnet-bluetooth-le #41: Android - ConnectedDevice list not up to date

xabre, 2016-05-23:
Sometimes a inconssistency appear between the internal list of connected devices and the truly connected devices. 

Use BluetyoothManager:
public IList<BluetoothDevice> GetConnectedDevices ([GeneratedEnum] ProfileType profile);

to try to detect inconsistencies. 

Scenario: 
ConnectedDevice list is empty. The real BLE device is still connected.

smstuebe, 2016-05-23:
Maybe we should get rid of holding an internal list. If `RetrieveConnectedPeripherals` on iOS and `GetConnectedDevices` on Android are not consuming alot of time, we should use them, because the OS knows the truth (hopefully).

domo4all, 2016-06-13:
iOS and Android already implement a cache service. As @smstuebe propose, why not leave the system do its job?

xabre, 2016-06-28:
I will move this issue to the 1.1.0 release.

xabre, 2016-09-19:
The problem with these methods is that they return system wide connected devices (includes devices connected to other apps)...