dotnet-bluetooth-le/dotnet-bluetooth-le #42: Get the connected device(s) from IAdapter.ConnectToDeviceAsync

Cheesebaron, 2016-05-24:
ConnectToDeviceAsync among other async methods do not return anything other than a Task. This is really confusing when dealing with async methods as you would, especially on ConnectToDeviceAsync, expect it to return the connected device or devices. Currently you have to hook up an event and wait for them to appear there, which is not really super nice.

Let me explain with some code. Say you have device `Sensor` you want to connect to and discover some services and characteristics and maybe do some reading and writing to said characteristics.

Currently you would have to do something like:

```
bleAdapter.DeviceConnected += BleAdapterOnDeviceConnected;

await bleAdapter.ConnectToDeviceAsync(sensor);
```

Then in the method `BleAdapterOnDeviceConnected` you would get notified about connected devices, so you go await from the scope where you called await on ConnectToDeviceAsync and straight after you can't really know whether you have connected to a device yet or not, since you need to wait for the event to trigger.

Usually when I work with other libraries that have async stuff incorporated I would work something like this:

```
IDevice device = await bleAdapter.ConnectToDeviceAsync(sensor);
// do stuff with device
```

Maybe even if we _need_ to disconnect from the device automatically make a `IDisposable` implementation:

```
using(var device = await bleAdapter.ConnectToDeviceAsync(sensor))
{
    // do stuff with connected device
}

// device is disconnected here
```

Would be really awesome if this library could support something like this.

smstuebe, 2016-05-24:
in 1.0.0 you don't have to use the event. I wrote it in the blog post (we have to update the readme). Maybe it's not clear enough, because we didn't provide the xml documentation with the nuget package.

	/// <summary>
	/// Connects to the <paramref name="device"/>.
	/// </summary>
	/// <param name="device">Device to connect to.</param>
	/// <param name="autoconnect">Android only: Automatically try to reconnect to the device, after the connection got lost. The default is false.</param>
	/// <param name="cancellationToken">The token to monitor for cancellation requests. The default value is None.</param>
	/// <returns>A task that represents the asynchronous read operation. The Task will finish after the device has been connected successfuly.</returns>
	/// <exception cref="DeviceConnectionException">Thrown if the device connection fails.</exception>
	Task ConnectToDeviceAync(IDevice device, bool autoconnect = false, CancellationToken cancellationToken = default(CancellationToken));

something like 

	try{
		await _adapter.ConnectToDeviceAync(device);
	}
	catch(DeviceConnectionException e){
	
	}

should be possible, with 1.0.0 already

Cheesebaron, 2016-05-24:
Alright thanks! Changed my code and this seems to work fine.

closed 2016-05-24 by Cheesebaron