dotnet-bluetooth-le/dotnet-bluetooth-le #43: Disconnect / Close

smstuebe, 2016-05-24:
Just have seen this on Android: http://stackoverflow.com/questions/25330938/android-bluetoothgatt-status-133-register-callback Just double check it (don't fully trust the answer, read the docs)
TODO:

- [ ] what we do on disconnect?
- [ ] Is there an Issue?

adolgov, 2016-05-24:
This is a particularly tricky issue because it seems to vary greatly with different phones/manufacturers (since they tend to provide their own bluetooth stack implementations).

133 is an odd generic error which seems, in my experience, to be from random zombie gatt stuff piling up. The reason why GattCallback catches errors (in my experience 133 is the one that pops up often, especially on a Samsung Galaxy S5) is to formally try to close out any connections and gatts. If we're not meticulous at that, I've seen the phone get into a state where it continuously gets 133 errors. At that point, closing the program might not even clear out the stale zombie gatts; sometimes turning bluetooth off-then-on fixes it, but other times I've had to do a full phone power cycle to resolve.

Another thing with Android is that it's fickle when it comes to timing. The calls to the gatt, even after the callback is received, only signify that your command/intention was passed on, not that it was actually fully executed. I've seen that if you follow up with other gatt calls too quickly, you can catch the OS in a weird transitional state (working on a close request, but not fully closed/disposed yet, and now a new open/read/whatever call comes in on a connection/gatt whose state is not truly known), and this can lead to unreliable behavior.

Being _very_meticulous about connection management and closing everything out every time, and introducing delays between successive calls to connection-establishing/closing functions seems to have pretty reliable behavior for me; however, the fact that behavior seems to vary significantly with different phone models, and the overall structure of Android's implementation of BLE still makes me nervous. It just seems like a very poor design for implementations that open and close connections on a regular basis.

Another variable that complicates things, in my experience, is whether a connection is closed cleanly by us, closed by the other end, or (most error-prone case) broken uncleanly from the other end (e.g. loss of power).

Anyhow, these are my experiences from dealing with still stuff and putting in various edge-case handlers when I was more actively working with this three or so months ago.

allbdeveloper, 2016-09-10:
I'm having this got gatt-133 error crash issue many times on Samsung Galaxy s4.

ccanbek, 2017-07-31:
I can confirm that this happens on samsung J7 in version 1.3 pretty consistently.