dotnet-bluetooth-le/dotnet-bluetooth-le #44: Question: Plugin BLE for Android

alberk8, 2016-05-27:
I am interested in your Plugin BLE for Android.   I would really help me to understand if there is sample app.  Thank you.

xabre, 2016-05-27:
@alberk8 Yes you can find the sample app in the solution (subfolder). [Client](https://github.com/xabre/xamarin-bluetooth-le/tree/master/Source/BLE.Client)

closed 2016-05-27 by xabre

alberk8, 2016-05-27:
Hi,

The sample is based on the MvvmCross.  What I need is just plain vanilla Android sample.  Thank you.

xabre, 2016-05-27:
It works the same. The only difference is that instead of resolving the IBluetoothLE interface using MvvmCross dependecy injection you do it statically by using **CrossBluetoothLE.Current**. Everything else works the same.

alberk8, 2016-05-27:
Sorry for bothering.  I am a newbie in c# and Xamarin so a sample app on Android would be very helpful.   Thank you again.

smstuebe, 2016-05-27:
The sample app is for Android and iOS. We won't provide a pure Android example (without Xamarin.Forms) project because this is too much effort for maintenance. The plugin works independent from the used UI technology. 

Take some time and analyze the sample well. But keep in mind that `IAdapter` isn't 'magically' injected by MvvmCross and you have to use `CrossBluetoothLE.Current.Adapter` to get it. If you have concrete questions, just ask them.