dotnet-bluetooth-le/dotnet-bluetooth-le #46: Vanilla Android ListView Adapters

LzJustin, 2016-06-02:
I'm trying to use the non-mvvm plugin for a pure android application that has a ListView to store the Devices result. So I do not know what is the right way to do it. Do I need to create a listview adapter to store the device? The sample app has Device, device, Devices and so on. So it's confusing for a beginner like me who would like to just have a simple click button to discover device.

My code so far to display result is

```
public IDevice Device { get; } //In my MainActivity class

//on my OnCreate(Bundle bundle)
button.Click += ScanForDevices;
_bluetoothLe.Adapter.DeviceDiscovered += _bluetoothLe_DeviceDiscovred;
//

private void _bluetoothLe_DeviceDiscovred(object sender, DeviceEventArgs e)
        {
            var msg = string.Format(@"Device found: {0}
                     {1} - {2}", Device.Name, e.Device.Name, e.Device.Id, e.Device.Rssi);
            DisplayInformation(msg);
        }
                private void ScanForDevices(object sender, EventArgs e)
        {
            if (!_bluetoothLe.Adapter.IsScanning)
                _bluetoothLe.Adapter.StartScanningForDevicesAsync();
        }

```

Currently, even this simple showing device isn't working for me. It's working if I use the other plugin/way from this link  "https://elbruno.com/2015/09/08/vs2015-listing-bluetooth-le-devices-in-android-with-xamarin/"

Appreciate any help or advice given in advance.

smstuebe, 2016-06-02:
can you change it to: 

    var msg = string.Format(@"Device found: {0}
                     {1} - {2}", e.Device.Name, e.Device.Id, e.Device.Rssi);

??

I don't know why you added a `Device` property. If you just scan, all information come from the `DeviceEventArgs e`.

LzJustin, 2016-06-02:
Yes I've changed it. But I wanted to see the differences of the output of both Device.Name and e.Device.Name.

But the point is, there is still no output coming up. I'm very sure my DisplayInformation method is not the issue. I must be missing something. How is it that I can add the Device.Name into a listview? I'm currently trying to put the string msg inside my list<string>. listDevice.add(msg); <- Hopefully this works.

smstuebe, 2016-06-02:
As I said... I don't know what your `Device` property is. If it's null, you get a null reference exception. Maybe you have to wrap the add with `RunOnUiThread();`

    RunOnUiThread(() => DisplayInformation(msg));

Or does it not enter your `_bluetoothLe_DeviceDiscovred` Do you know what a debugger is? "It does not work" is not very helpful.

xabre, 2016-06-03:
Maybe if you post your entire MainActivity I could help you more. I don't know how your DisplayInformation method works. Did you try Console.WriteLine(msg) ? :)

LzJustin, 2016-06-03:
Xabre, Okay, I've posted the whole MainActivity below. Thank you very much in advance. And I'll try to do a Console.WriteLine(msg) now instead of (line). 

Smstuebe, sorry, I am not very familiar with coding yet. I'm still desperately trying to learn. I've tried the debugger, it has no Null Reference exception. I moused over Device and it is (field) IDevice DeviceEventArgs.Device. By property of the Device, does it mean I should actually create a "Device" class and set the properties? Cause so far mousing over the Name/Id/Rssi is "string IDevice.Name { get;}

     public class MainActivity : Activity
          {
         //ListView mListView;
         private EditText _textboxResults;
         private IBluetoothLE _bluetoothLe;
         private Button button;
         private Plugin.BLE.Abstractions.Contracts.IAdapter adapter;
         private ListView deviceList;

        List<string> lol2 = new List<string>();

        protected override void OnCreate(Bundle bundle)
        {
            base.OnCreate(bundle);

            _bluetoothLe = CrossBluetoothLE.Current;
            adapter = CrossBluetoothLE.Current.Adapter;
            var state = _bluetoothLe.State;

            // Set our view from the "main" layout resource
            SetContentView(Resource.Layout.Main);

            button = FindViewById<Button>(Resource.Id.MyButton);
            _textboxResults = FindViewById<EditText>(Resource.Id.editText1);
            ListView deviceList = FindViewById<ListView>(Resource.Id.listView1);

            button.Click += ScanForDevices;
            _bluetoothLe.Adapter.DeviceDiscovered += _bluetoothLe_DeviceDiscovred;

        }

        private void _bluetoothLe_DeviceDiscovred(object sender, DeviceEventArgs e)
        {
            var msg = string.Format(@"Device found: {0}
                     {1} - {2}", e.Device.Name, e.Device.Id, e.Device.Rssi);
            lol2.Add(msg);
            RunOnUiThread(() => DisplayInformation(msg));
        }

        private void ScanForDevices(object sender, EventArgs e)
        {
            
                _bluetoothLe.Adapter.StartScanningForDevicesAsync();

        }
        private void DisplayInformation(string line)
        {
            _textboxResults.Text = $"{line}\r\n{_textboxResults.Text}";
            Console.WriteLine(line);
        }

xabre, 2016-06-03:
I'm investigating this now. There seems to be a problem with the Plugin.Ble.dll packed in the nuget package... the one built from source works fine. I'll keep you updated.

xabre, 2016-06-03:
Until I can pinpoint the packaging problem, I've tried version 1.0.0-beta with your code and it works fine. For now try to install :

`Install-Package Plugin.BLE -Version 1.0.0-beta -Pre`

This code works with the older version:
`public class MainActivity : Activity
    {
        //ListView mListView;
        private EditText _textboxResults;
        private IBluetoothLE _bluetoothLe;
        private Button _button;

        protected override void OnCreate(Bundle bundle)
        {
            base.OnCreate(bundle);

            _bluetoothLe = CrossBle.Current;

            // Set our view from the "main" layout resource
            SetContentView(Resource.Layout.Main);

            _button = FindViewById<Button>(Resource.Id.MyButton);
            _textboxResults = FindViewById<EditText>(Resource.Id.editText1);
            //ListView deviceList = FindViewById<ListView>(Resource.Id.listView1);

            _button.Click += ScanForDevices;
            _bluetoothLe.Adapter.DeviceDiscovered += _bluetoothLe_DeviceDiscovred;
        }

        private void _bluetoothLe_DeviceDiscovred(object sender, DeviceEventArgs e)
        {
            var msg = $@"Device found: {e.Device.Name}
                 {e.Device.Id} - {e.Device.Rssi}";

            RunOnUiThread(() => DisplayInformation(msg));
        }

        private void ScanForDevices(object sender, EventArgs e)
        {
            _bluetoothLe.Adapter.StartScanningForDevicesAsync();

        }
        private void DisplayInformation(string line)
        {
            Console.WriteLine(line);
            _textboxResults.Text = $"{line}\r\n{_textboxResults.Text}";

        }
    }`

I'll try to fix the new package and then release an update and keep you up to date.

xabre, 2016-06-03:
@LzJustin you should be able to update to version 1.0.0-beta3 now. Rebuilding the nuget package fixed the problem.

LzJustin, 2016-06-06:
@xabre , thank you for updating it. I've updated to the newest beta3 but the application still fails to discover device at my side. So to test out whether the adapter is really scanning or not. I tried this

    public class MainActivity : Activity
    {
        //ListView mListView;
        private EditText _textboxResults;
        private Button button;
        private ListView deviceList;
        private IBluetoothLE _bluetoothLe;
        private Plugin.BLE.Abstractions.Contracts.IAdapter adapter;
        //List<DeviceListItemViewModel> lol = new List<DeviceListItemViewModel>();
        List<string> lol2 = new List<string>();
        private EditText _textboxResults2;

        protected override void OnCreate(Bundle bundle)
        {
            base.OnCreate(bundle);

            // Set our view from the "main" layout resource
            SetContentView(Resource.Layout.Main);

            adapter = CrossBluetoothLE.Current.Adapter;
            _bluetoothLe = CrossBluetoothLE.Current;

            button = FindViewById<Button>(Resource.Id.MyButton);
            _textboxResults = FindViewById<EditText>(Resource.Id.editText1);
            _textboxResults2 = FindViewById<EditText>(Resource.Id.editText2);
            // deviceList = FindViewById<ListView>(Resource.Id.listView1);
            button.Click += ScanForDevices;

            _bluetoothLe.Adapter.DeviceDiscovered += Simplemessage;
            //_bluetoothLe.DeviceDiscovered += _bluetoothLe_DeviceDiscovred;
        }

        private void Simplemessage(object sender, DeviceEventArgs e)
        {
            var line = "";
            line = string.Format("discovered");
            RunOnUiThread(() => DisplayInformation2(line));
        }

        private void ScanForDevices(object sender, EventArgs e)
        {
            var line = "";

            if (!adapter.IsScanning)
            {
                line = string.Format("Not Scanning");
                adapter.StartScanningForDevicesAsync();
            }
            else
            {
                line = string.Format("Is Scanning");
                _bluetoothLe.Adapter.DeviceDiscovered += Simplemessage;

            }
            RunOnUiThread(() => DisplayInformation(line));
            //_bluetoothLe.StartScanningForDevicesAsync();
            
        }
        private void DisplayInformation(string msg)
        {
            _textboxResults.Text = $"{msg}\r\n{_textboxResults.Text}";
            Console.WriteLine(msg);
        }
        private void DisplayInformation2(string line)
        {
            _textboxResults2.Text = $"{line}\r\n{_textboxResults2.Text}";
            Console.WriteLine(line);
        }

I ran this, on the first tap on the button. It's natural to show "Not Scanning". Then on the second tap, it shows "Is Scanning", which is what I want. Then I tested it by turning off my Bluetooth on my Phone(Exported the APK to run on physical device), and surprisingly it still shows "is scanning" on the second tap. Also, the  `_bluetoothLe.Adapter.DeviceDiscovered` doesn't seem to trigger the Simplemessage method. Any guidance?

LzJustin, 2016-06-06:
@xabre , I've fixed it, might be something wrong with my project. I re-did everything similar to the codes you gave and now it's working properly. I may have messed up the previous project. But thank you for updating and also the help so far :) 

@smstuebe , thank you too for the help :)

closed 2016-06-06 by smstuebe