dotnet-bluetooth-le/dotnet-bluetooth-le #47: Connect to known device without discovering it before.

domo4all, 2016-06-06:
Hi,
First of all, it is a very cool project. Thank you very much for the good work.

I tried the library for some days and I wonder if there is a way to connect to a known device (by its GUID) directly, without performing a long discovery before?

Also, I noticed that if a disconnect from a device, than I can connect again to that device but cannot exchange any data. I have to perform a discovery before to make it work again.
Do I make something wrong?

Thank you for your answers.
Cheers.

smstuebe, 2016-06-06:
Hi,

1. issue: I have never thought about this scenario. We have to investigate it, unless @xabre knows any reasons against without thinking about it.
2. issue: iOS or Android? Or both?

domo4all, 2016-06-07:
Hi,

Thank you for the quick reply.

1. I think it could be a good approach to speed up connections with BLE devices. Here is what I've planned: Make a first discovery in the app and store BLE devices informations in a database (ie. device GUID) . Then, use these data to make a quick connect to a specific device to perform an action (ie. turn on a light). Finally, disconnect from device.
I've heard of something like a "cache" in OS. Don't known if it could help or not to speed up connections.

2. I'm making my test with an official Nexus 5 (first gen). My solution (in Xamarin Beta channel) is a pure MvvmCross one with a Core and a Droid projects in it (not Forms). I think I'm facing the same problem with the BLE Demo project on my iPhone 6S. But need to check again.

xabre, 2016-06-07:
1. We used to have a Extension method name something like "ConnectToSpecificDeviceAsync(Guid guid)" which was actually discovering the specific device and then connecting to it and returning after these two steps were complete.
Both Android and iOS need a 'BluetoothDevice' or a 'CBPeripherial' in order to connect and these objects are instantiated by the OS specific API's on discovery. I'm not sure if it's possible to skip the discovery phase. More investigation is required.
2. Can you describe the specific steps you took to reproduce the problem?

xabre, 2016-06-07:
@domo4all I think I know what the problem is. We do not clear the list of KnownServices on disconnect, and when reusing the instance of the IDevice it uses the list of cached services from the previous connection.

domo4all, 2016-06-07:
@xabre Sounds like it could be the problem. I effectively reuse the instance of IDevice I got from an original and unique discovery. I will check this at home and paste here some logs.

domo4all, 2016-06-07:
@xabre 
OK, here are some logs. Hope it will help.
As you see, the second connection is successful but StartUpdates return False. No data can transmit.

---> Discovery

Adapter >= 21: Starting a scan for devices.
[BluetoothAdapter] STATE_ON
[BluetoothLeScanner] onClientRegistered() - status=0 clientIf=5
Adv rec [Type Flags; Data 06]
Adv rec [Type ManufacturerSpecificData; Data 48-4D-20-91-48-BB-C8-9A]
Adv rec [Type ServiceData; Data 00-B0-00-00-00-00]
Adv rec [Type UuidsIncomple16Bit; Data FF-E0]
Adv rec [Type CompleteLocalName; Data 48-4D-53-6F-66-74-00-00-00-00-00-00-00]
Adv rec [Type TxPowerLevel; Data 00]
Adv rec [Type Flags; Data 06]
Adv rec [Type ManufacturerSpecificData; Data 48-4D-20-91-48-BB-C8-9A]
Adv rec [Type ServiceData; Data 00-B0-00-00-00-00]
Adv rec [Type UuidsIncomple16Bit; Data FF-E0]
Adv rec [Type CompleteLocalName; Data 48-4D-53-6F-66-74-00-00-00-00-00-00-00]
Adv rec [Type TxPowerLevel; Data 00]
Adv rec [Type Flags; Data 06]
Adv rec [Type ManufacturerSpecificData; Data 48-4D-20-91-48-BB-C8-9A]
Adv rec [Type ServiceData; Data 00-B0-00-00-00-00]
Adv rec [Type UuidsIncomple16Bit; Data FF-E0]
Adv rec [Type CompleteLocalName; Data 48-4D-53-6F-66-74-00-00-00-00-00-00-00]
Adv rec [Type TxPowerLevel; Data 00]
Adv rec [Type Flags; Data 06]
Adv rec [Type ManufacturerSpecificData; Data 48-4D-20-91-48-BB-C8-9A]
Adv rec [Type ServiceData; Data 00-B0-00-00-00-00]
Adv rec [Type UuidsIncomple16Bit; Data FF-E0]
Adv rec [Type CompleteLocalName; Data 48-4D-53-6F-66-74-00-00-00-00-00-00-00]
Adv rec [Type TxPowerLevel; Data 00]
Adv rec [Type Flags; Data 06]
Adv rec [Type ManufacturerSpecificData; Data 48-4D-20-91-48-BB-C8-9A]
Adv rec [Type ServiceData; Data 00-B0-00-00-00-00]
Adv rec [Type UuidsIncomple16Bit; Data FF-E0]
Adv rec [Type CompleteLocalName; Data 48-4D-53-6F-66-74-00-00-00-00-00-00-00]
Adv rec [Type TxPowerLevel; Data 00]
Adv rec [Type Flags; Data 06]
Adv rec [Type ManufacturerSpecificData; Data 48-4D-20-91-48-BB-C8-9A]
Adv rec [Type ServiceData; Data 00-B0-00-00-00-00]
Adv rec [Type UuidsIncomple16Bit; Data FF-E0]
Adv rec [Type CompleteLocalName; Data 48-4D-53-6F-66-74-00-00-00-00-00-00-00]
Adv rec [Type TxPowerLevel; Data 00]
Adv rec [Type Flags; Data 06]
Adv rec [Type ManufacturerSpecificData; Data 48-4D-20-91-48-BB-C8-9A]
Adv rec [Type ServiceData; Data 00-B0-00-00-00-00]
Adv rec [Type UuidsIncomple16Bit; Data FF-E0]
Adv rec [Type CompleteLocalName; Data 48-4D-53-6F-66-74-00-00-00-00-00-00-00]
Adv rec [Type TxPowerLevel; Data 00]
Adv rec [Type Flags; Data 06]
Adv rec [Type ManufacturerSpecificData; Data 48-4D-20-91-48-BB-C8-9A]
Adv rec [Type ServiceData; Data 00-B0-00-00-00-00]
Adv rec [Type UuidsIncomple16Bit; Data FF-E0]
Adv rec [Type CompleteLocalName; Data 48-4D-53-6F-66-74-00-00-00-00-00-00-00]
Adv rec [Type TxPowerLevel; Data 00]
Adv rec [Type Flags; Data 06]
Adv rec [Type ManufacturerSpecificData; Data 48-4D-20-91-48-BB-C8-9A]
Adv rec [Type ServiceData; Data 00-B0-00-00-00-00]
Adv rec [Type UuidsIncomple16Bit; Data FF-E0]
Adv rec [Type CompleteLocalName; Data 48-4D-53-6F-66-74-00-00-00-00-00-00-00]
Adv rec [Type TxPowerLevel; Data 00]
Adv rec [Type Flags; Data 06]
Adv rec [Type ManufacturerSpecificData; Data 48-4D-20-91-48-BB-C8-9A]
Adv rec [Type ServiceData; Data 00-B0-00-00-00-00]
Adv rec [Type UuidsIncomple16Bit; Data FF-E0]
Adv rec [Type CompleteLocalName; Data 48-4D-53-6F-66-74-00-00-00-00-00-00-00]
Adv rec [Type TxPowerLevel; Data 00]
Adapter: Scan timeout has elapsed.
Adapter: Stopping the scan for devices.
Adapter >= 21: Stopping the scan for devices.
[BluetoothAdapter] STATE_ON

---> Connection

[BluetoothManager] getConnectionState()
[BluetoothManager] getConnectedDevices
[BluetoothGatt] connect() - device: 20:91:48:BB:C8:9A, auto: false
[BluetoothGatt] registerApp()
[BluetoothGatt] registerApp() - UUID=2d33fbef-fc0a-4198-92b3-ec56940d9a15
[BluetoothGatt] onClientRegistered() - status=0 clientIf=5
[BluetoothGatt] onClientConnectionState() - status=0 clientIf=5 device=20:91:48:BB:C8:9A
GattCallback state: Connected
Connected
ConnectToDeviceAync Connected: 00000000-0000-0000-0000-209148bbc89a HMSoft
[BluetoothGatt] discoverServices() - device: 20:91:48:BB:C8:9A
[BluetoothGatt] onSearchComplete() = Device=20:91:48:BB:C8:9A Status=0
OnServicesDiscovered: Success
Characteristic.StartUpdates
[BluetoothGatt] setCharacteristicNotification() - uuid: 0000ffe1-0000-1000-8000-00805f9b34fb enable: true
Characteristic.StartUpdates, successful: True

---> Disconnect

[BluetoothGatt] cancelOpen() - device: 20:91:48:BB:C8:9A
[BluetoothGatt] onClientConnectionState() - status=0 clientIf=5 device=20:91:48:BB:C8:9A
GattCallback state: Disconnected
Disconnected by user
[BluetoothGatt] close()
[BluetoothGatt] unregisterApp() - mClientIf=5
DisconnectedPeripheral by user: HMSoft
DisconnectAsync Disconnected: 00000000-0000-0000-0000-209148bbc89a HMSoft

---> Connection

[BluetoothManager] getConnectionState()
[BluetoothManager] getConnectedDevices
[BluetoothGatt] connect() - device: 20:91:48:BB:C8:9A, auto: false
[BluetoothGatt] registerApp()
[BluetoothGatt] registerApp() - UUID=fe28ae36-aded-4c3d-8e09-532b1788484a
[BluetoothGatt] onClientRegistered() - status=0 clientIf=5
[BluetoothGatt] onClientConnectionState() - status=0 clientIf=5 device=20:91:48:BB:C8:9A
GattCallback state: Connected
Connected
ConnectToDeviceAync Connected: 00000000-0000-0000-0000-209148bbc89a HMSoft
Characteristic.StartUpdates
[BluetoothGatt] setCharacteristicNotification() - uuid: 0000ffe1-0000-1000-8000-00805f9b34fb enable: true
Characteristic.StartUpdates, successful: False

xabre, 2016-06-08:
Yes, there is no call to discover services after the second connection
`[BluetoothGatt] discoverServices() - device: 20:91:48:BB:C8:9A`

We will publish a fix for this problem. Maybe later today -beta4.

domo4all, 2016-06-08:
Yes. That's the point.
Thank you for the fix. I will keep testing it.

domo4all, 2016-06-08:
@xabre For my first question on how to reconnect to a known device by its UUID, I just saw that it is possible on iOS (not checked on Android). And that's very cool and exactly what I expect.

Please have a look at this video and jump to 12"50. It is well explained:
https://developer.apple.com/videos/play/wwdc2012/705/

Cheers.

smstuebe, 2016-06-08:
I split up the issue into a feature request (this issue) and a bug (#48) So we can handle them separately.

yoldasmehmed, 2016-06-12:
Is there a sample code for to connect specific device.
I connect specific device on android like the code belows,
 Java.Util.UUID serialUUID = Java.Util.UUID.FromString("00001101-0000-1000-800000805F9B34FB");
BluetoothDevice bluetoothDevice = bluetoothAdapter.GetRemoteDevice(mac);
 bluetoothSocket = bluetoothDevice.CreateRfcommSocketToServiceRecord(serialUUID);

How can I change this code for portable platform using this library.

Best regards..

xabre, 2016-06-13:
Currently the plugin only supports Bluetooth LE 4.0. BluetootkSocket is
part of the bl 2.1 api which is not supported at the moment, only
BluetoothGatt
<https://developer.android.com/reference/android/bluetooth/BluetoothGatt.html>
is
supported.

xabre, 2016-10-03:
Check out 1.1.0-beta*

closed 2016-10-03 by xabre