dotnet-bluetooth-le/dotnet-bluetooth-le #48: Clear cached services after disconnect.

smstuebe, 2016-06-08:
Pulled the bug out from issue #47 so we can handle feature request and bug separately

from @domo4all:

Also, I noticed that if I disconnect from a device, than I can connect again to that device but cannot exchange any data. I have to perform a discovery before to make it work again.
Do I make something wrong?

from @xabre:
I think I know what the problem is. We do not clear the list of KnownServices on disconnect, and when reusing the instance of the IDevice it uses the list of cached services from the previous connection.

Yes, there is no call to discover services after the second connection
[BluetoothGatt] discoverServices() - device: 20:91:48:BB:C8:9A
We will publish a fix for this problem. Maybe later today -beta4. Thank you for pointing it out.

xabre, 2016-06-10:
@domo4all We have just released 1.0.0-beta4 which should include the fix for this issue.

domo4all, 2016-06-10:
@xabre Thank you. I tried it and it works perfectly. Really nice job :)

closed 2016-06-10 by xabre