dotnet-bluetooth-le/dotnet-bluetooth-le #49: Retrieving Services after Selecting Device(Android)

LzJustin, 2016-06-09:
Unable to display Services from the selected Device (on my first page ListView).

var services = await connectedDevice.GetServicesAsync();

This line here is the example given on Readme. How do I create the _connectedDevice object which can refer back to what I have selected on my previous Activity? It gives off a NullException error when trying to debug it. Which is crashing at GetServicesAsync.

Here are my codes on my ServiceList.cs. If needed, please check attached file for the ServiceListAdapter (It's abit long if I put them all inside this issue)
[ServiceListAdapter.txt](https://github.com/xabre/xamarin-bluetooth-le/files/306336/ServiceListAdapter.txt)

        private ListView servicelistview;
        private IDevice _connectedDevice;
        public IList<IService> _services;
        private TextView tv1;
        private IService item;
        private ServiceListAdapter sla;

        protected override void OnCreate(Bundle savedInstanceState)
        {
            base.OnCreate(savedInstanceState);
            //var selectedItemId = Intent.Extras.GetString("DeviceSelected");
            SetContentView(Resource.Layout.servicelist);

            tv1 = FindViewById<TextView>(Resource.Id.textView1);

           // tv1.Text = selectedItemId;

            servicelistview = FindViewById<ListView>(Resource.Id.listView1);

            getServices();
        }

        private async void getServices()
        {
            if (_services == null)
            {
                tv1.Text = "Services is null";
                _services = await _connectedDevice.GetServicesAsync();
                showServices();
            }

            else
            {
                tv1.Text = "Show Services";
                showServices();
            }

        }

        private void showServices()
        {
            //_services.Add(item);
            sla = new ServiceListAdapter(this, _services);
            servicelistview.Adapter = sla;

            sla.NotifyDataSetChanged();
        }

smstuebe, 2016-06-09:
Where do you set `_connectedDevice` ??

LzJustin, 2016-06-09:
This is actually another activity after selecting a Device Item from my MainActivity class. So I don't think I've done anything with _connectedDevice other than declaring that IDevice object in the class ServiceList. What am I suppose to set for the _connectedDevice?

smstuebe, 2016-06-09:
Hmm this is a programming issue. We cant teach you programming :/ Could you please post this question on Stackoverflow?

LzJustin, 2016-06-09:
I've tried asking questions on Stackoverflow before but the amount of Xamarin Android users are way too little. Not to mention about something so specified like using this Plugin to retrieve the Services. But I'll try my luck there too I guess. Thanks :) !

xabre, 2016-06-09:
Simplest dirty way is to make something like a singleton or public static class with a static IDevice property which you set in the first activity after connecting and then read in the second.

LzJustin, 2016-06-09:
I see, I'll check what I can come out with using the Static IDevice. Hopefully it can show the services since I'm actually planning to do like the CrossPlatform way, to be able to connect to multiple devices (and hopefully it wouldn't overwrite each other). Thanks @xabre ! You've helped plenty :) !.

LzJustin, 2016-06-10:
Found the solution to this, hope it helps other users that want to use the vanilla plugin instead.

Referring to the code below, simply putExtra in the DeviceList activity and pass the Position of the Device selected to the serviceList activity in order to know which device you're on and also getting the IDevice object. Then the IDevice in the serviceListActivity will be the one connected and also able to call its GetServicesAsync method.

      base.OnCreate(savedInstanceState);
            var selectedItemId = Intent.Extras.GetInt("DeviceSelected");
            SetContentView(Resource.Layout.servicelist);

            _adapter = CrossBluetoothLE.Current.Adapter;

            var connected = _adapter.ConnectedDevices[selectedItemId];

            tv1 = FindViewById<TextView>(Resource.Id.textView1);

            servicelistview = FindViewById<ListView>(Resource.Id.listView1);

            _services = await connected.GetServicesAsync();

            sla = new ServiceListAdapter(this, _services);
            servicelistview.Adapter = sla;

            sla.NotifyDataSetChanged();

closed 2016-06-10 by smstuebe