dotnet-bluetooth-le/dotnet-bluetooth-le #50: Acting as a peripheral

domo4all, 2016-06-13:
Hello,
I was wondering if you planned to manage Peripheral mode, so that a smartphone or tablet can send advertisements, can be connected from another BLE devices, etc.

Cheers.

smstuebe, 2016-06-13:
Yes, https://github.com/xabre/xamarin-bluetooth-le/tree/ble-server 

But we focus on a neat client API, in first place.

xabre, 2016-06-13:
Actually we have a '**server**' or '**peripheral**' package planned. It's currently in development on a different branch (issue #29 :)). Will keep you up to date with it's progress.

domo4all, 2016-06-13:
Oh. Nice. I missed the other branches. Very cool. Thanks guys.

closed 2016-06-13 by smstuebe

ChrisMash, 2017-12-01:
Just wondering what the state of the ble-server branch is? I'm hoping to connect a few iOS/Android devices together with one as a host and the others as peripherals but it kinda seems like it's not going to be possible with this plugin and the ble-server hasn't been touched for two years, with no code on the iOS side?