dotnet-bluetooth-le/dotnet-bluetooth-le #51: Error XA2006 Referencing back to Plugin.BLE

LzJustin, 2016-06-15:
Hi again,

I've come across this error. On the Plugin.BLE beta3, I can release the solution and build an APK. But if I updated my package to beta4 and try to release the solution. This will pop up. Any clue? I do need to use the beta4 for now. Can't stick to beta3 for too long.

Severity	Code	Description	Project	File	Line	Suppression State
Error		The "LinkAssemblies" task failed unexpectedly.
Xamarin.Android.XamarinAndroidException: error XA2006: Reference to metadata item '!!0[] System.Array::Empty()' (defined in 'Plugin.BLE, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null') from 'Plugin.BLE, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null' could not be resolved. ---> Mono.Cecil.ResolutionException: Failed to resolve !!0[] System.Array::Empty()
   at Mono.Linker.Steps.MarkStep.MarkMethod(MethodReference reference)
   at Mono.Linker.Steps.MarkStep.MarkInstruction(Instruction instruction)
   at Mono.Linker.Steps.MarkStep.MarkMethodBody(MethodBody body)
   at Mono.Linker.Steps.MarkStep.ProcessMethod(MethodDefinition method)
   at Mono.Linker.Steps.MarkStep.ProcessQueue()
   at Mono.Linker.Steps.MarkStep.Process()
   at Mono.Linker.Steps.MarkStep.Process(LinkContext context)
   at Mono.Linker.Pipeline.Process(LinkContext context)
   at MonoDroid.Tuner.Linker.Process(LinkerOptions options, LinkContext& context)
   at Xamarin.Android.Tasks.LinkAssemblies.Execute()
   --- End of inner exception stack trace ---
   at Xamarin.Android.Diagnostic.Error(Int32 code, Exception innerException, String message, Object[] args)
   at Xamarin.Android.Tasks.LinkAssemblies.Execute()
   at Microsoft.Build.BackEnd.TaskExecutionHost.Microsoft.Build.BackEnd.ITaskExecutionHost.Execute()
   at Microsoft.Build.BackEnd.TaskBuilder.<ExecuteInstantiatedTask>d__26.MoveNext()	App2

smstuebe, 2016-06-15:
Have you installed the latest stable Xamarin Version?

LzJustin, 2016-06-16:
Just installed it. No link assemblies problem anymore. Thank you :+1:  !

closed 2016-06-16 by smstuebe