dotnet-bluetooth-le/dotnet-bluetooth-le #52: error with nuget

sebastienjouhans, 2016-06-15:
I tried to install the nuget package via the package manager but when I am trying to do so I get the following error

Install failed. Rolling back...
Package 'Plugin.BLE.1.0.0-beta4' does not exist in project 'BluetoothToggleWidget'
Package 'Plugin.BLE.1.0.0-beta4' does not exist in folder 'C:\Users\sebastienj\Documents\WORK\MyStuff\XamarinBLETest\Xamarin\AndroidChat\packages'
Install-Package : Failed to add reference to 'Plugin.BLE.Abstractions'.
At line:1 char:1
+ Install-Package Plugin.BLE -Pre
+ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : NotSpecified: (:) [Install-Package], Exception
    + FullyQualifiedErrorId : NuGetCmdletUnhandledException,NuGet.PackageManagement.PowerShellCmdlets.InstallPackageCommand
 

Can you please fix it?

smstuebe, 2016-06-15:
> Attempting to gather dependency information for package 'Plugin.BLE.1.0.0-beta4' with respect to project 'App1', targeting 'MonoAndroid,Version=v6.0.99'
Attempting to resolve dependencies for package 'Plugin.BLE.1.0.0-beta4' with DependencyBehavior 'Lowest'
Resolving actions to install package 'Plugin.BLE.1.0.0-beta4'
Resolved actions to install package 'Plugin.BLE.1.0.0-beta4'
  GET https://www.nuget.org/api/v2/package/Plugin.BLE/1.0.0-beta4
  OK https://www.nuget.org/api/v2/package/Plugin.BLE/1.0.0-beta4 1276ms
Installing Plugin.BLE 1.0.0-beta4.
Adding package 'Plugin.BLE.1.0.0-beta4' to folder 'C:\Projekte\Temp\App1\packages'
Added package 'Plugin.BLE.1.0.0-beta4' to folder 'C:\Projekte\Temp\App1\packages'
Added package 'Plugin.BLE.1.0.0-beta4' to 'packages.config'
Successfully installed 'Plugin.BLE 1.0.0-beta4' to App1

It's working here. So you have to provide some more information. :)

xabre, 2016-07-29:
@sebastienjouhans What's the status of this issue? Is it still occuring for you?

xabre, 2016-08-02:
I'm guessing this has been solved. Will close.

closed 2016-08-02 by xabre