dotnet-bluetooth-le/dotnet-bluetooth-le #54: Xamarin.Mac support

wind-rider, 2016-06-18:
Interesting project! Are there any plans to support the macOS platform?

smstuebe, 2016-06-19:
Not yet, but nice idea. We are currently focusing on the 1.0 stable release. And have some feature ideas for later releases.

xabre, 2017-01-11:
ToDo:

- [ ] Create a functioning Xamarin.Mac collection view, with bindable item templates etc (or fix the current collection view which shows all items stacked on top of each other)
- [ ] Use this collection view for ServiceList, CharcateristicList and DescriptorList
- [ ] Finalize, unify Xamarin.Mac implementation of IUserDialogs
- [ ] Test all BLE functionality
- [ ] If appropriate, move the xamarin.mac and ios BLE implementation to a shared project and reference it from the platform implementations.

smstuebe, 2017-01-12:
Hmm can we use a shared project or file linking please? Or are there too many differences? At the fingerprint plugin I've done it, too.

flyingxu, 2017-05-30:
I hope it's not too late to suggest use Xamarin.Forms for Mac UI (https://blog.xamarin.com/preview-bringing-macos-to-xamarin-forms/). In that way, UI part will not be an issue and at least 3 more todo items in xabre's list might be marked as done.

xabre, 2019-08-04:
2.1.0-pre1

closed 2019-08-04 by xabre