dotnet-bluetooth-le/dotnet-bluetooth-le #55: Queries on 'Writing a value to a characteristic' in Plugin.BLE nuget package

leelatoo, 2016-06-20:
Hi,

I have a few queries related to 'Writing a value to a characteristic’ using Plugin.BLE nuget package. 
The write operation can be of two types: write with response and write without response.
 
According to Apple's documentation of Core Bluetooth framework, there are two types of characteristic write: CBCharacteristicWriteType - CBCharacteristicWriteWithResponse and CBCharacteristicWriteWithoutResponse. "The peripheral responds to write requests that are specified as CBCharacteristicWriteWithResponse by calling the peripheral:didWriteValueForCharacteristic:error: method of its delegate object."   
                                              
According to  Android BluetoothGatt documentation, we can set the write type for a characteristic. “Setting the write type of a characteristic determines how the writeCharacteristic(BluetoothGattCharacteristic) function write this characteristic.”. It can be one of: WRITE_TYPE_DEFAULT, WRITE_TYPE_NO_RESPONSE or WRITE_TYPE_SIGNED.

In Plugin.BLE nuget package, we could find there are a few characteristic property types defined related to write: AppleWriteWithoutResponse, WriteWithoutResponse,  AuthenticatedSignedWrites. 
And we have a method "WriteAsync(data)" to write a value to a characteristic, which returns a bool. If the return value is ‘true’, data is written successfully. If it is ‘false’, the data is not written to the characteristic.

1.	Is there any way to specify whether this is a writeWithResponse or writeWithoutResponse?  How to perform write with response using Plugin.BLE nugget package?

2.	What is the difference between WriteWithoutResponse and AppleWriteWithoutResponse?

Can you please help me getting answers for the above mentioned questions?

Thanks in advance :)

Regards,
Leela

smstuebe, 2016-06-20:
Hi,

this is related to #45.

1. No. We always assume, that the BLE peripheral tells us if the characteristic is with response or without.
2. No difference. I'll remove the Apple one by solving #45.

leelatoo, 2016-06-21:
Thanks for the answers :)  It helps to understand Plugin.BLE nuget package.

xabre, 2016-08-05:
Refer to #45 and #62.

closed 2016-08-05 by xabre