dotnet-bluetooth-le/dotnet-bluetooth-le #56: Characteristic Notify with acknowledgment aka. Indicate

xabre, 2016-06-29:
Currently we only support classical notifications (without ack). As pointed out by one of our collegues some devices require **indicate** instead of **notify**. Indicate is notify with ack.

For Android use:
```C#
descriptor.setValue(BluetoothGattDescriptor.ENABLE_INDICATION_VALUE)
```
instead of
```C#
descriptor.setValue(BluetoothGattDescriptor.ENABLE_NOTIFICATION_VALUE)
```

For iOS:
TBD

Dragon160, 2016-06-30:
Great work - I just tested indications on iOS and it seems to work!

xabre, 2016-06-30:
Happy to hear that :) Didi you get the chance to test it on Android?

Dragon160, 2016-06-30:
Not yet - I'll let you know

dweyermann, 2016-06-30:
First of all thanks very much for your great library. Your effort is really appreciated.

We currently use version 1.0.0-beta5 and have serious problems on a Samsung Galaxy Tab Active and Galaxy Xcover3, Android 5.1.1. iOS seems to be pretty reliable so far. Have you been able to get notifications reliably work on Android?

We communicate with a custom BLE device, and have a proprietary command/response protocol. Once we write a command to the "command" characteristic, we wait for the notification on the "response" characteristic (which contains the response of the device).

This works for several command/response sequences but suddenly notifications start to return invalid data. Sometimes a data byte in the response is missing, sometimes a data byte contains a invalid value. As mentioned this is only on Android.

xabre, 2016-07-07:
@dweyermann I have been using the plugin for similar use cases where we have a characteristic used only for notifications. So far we didn't notice any problems. Tested devices: Nexus 5 (5.1.1), Nexus 6 (6.0), Samsung Galaxy S5 (5.1.1), Oppo R7Plus (CM 6.0), Samasung Galaxy S3 Neo (4.4) ... It's probably best to open a separate issue for this and further investigate it.

dweyermann, 2016-07-13:
@xabre We have implemented to request a connection parameter update from the BLE device with appropriate parameters. We have not observed any problems with notifications since then.

Thanks for your help and time!

closed 2016-08-02 by xabre