dotnet-bluetooth-le/dotnet-bluetooth-le #57: Started Service Background Scanning

LzJustin, 2016-06-30:
Currently my started service and run and discover device when the application is "foreground". Which means it's not minimized. The output is fine

```
[0:] BLE Bluetooth is On
[0:] BLE IS SCANNING
06-30 15:38:24.396 D/Mono    (27664): Assembly Ref addref Plugin.BLE[0x7f9a6ab500] -> System.Core[0x7f7aae1180]: 4
Adapter >= 21: Starting a scan for devices.
06-30 15:38:24.404 I/mono-stdout(27664): Adapter >= 21: Starting a scan for devices.
06-30 15:38:24.417 D/BluetoothAdapter(27664): STATE_ON
06-30 15:38:24.429 D/BluetoothLeScanner(27664): onClientRegistered() - status=0 clientIf=6
06-30 15:38:24.454 D/Mono    (27664): [0x7f758a0440] hill climbing, change max number of threads 6
06-30 15:38:24.729 D/Mono    (27664): DllImport searching in: '__Internal' ('(null)').
06-30 15:38:24.730 D/Mono    (27664): Searching for 'java_interop_jnienv_is_assignable_from'.
06-30 15:38:24.730 D/Mono    (27664): Probing 'java_interop_jnienv_is_assignable_from'.
06-30 15:38:24.730 D/Mono    (27664): Found as 'java_interop_jnienv_is_assignable_from'.
06-30 15:38:24.735 D/Mono    (27664): DllImport searching in: '__Internal' ('(null)').
06-30 15:38:24.735 D/Mono    (27664): Searching for 'java_interop_jnienv_get_byte_array_region'.
06-30 15:38:24.735 D/Mono    (27664): Probing 'java_interop_jnienv_get_byte_array_region'.
06-30 15:38:24.735 D/Mono    (27664): Found as 'java_interop_jnienv_get_byte_array_region'.
06-30 15:38:24.766 D/Mono    (27664): Assembly Ref addref Plugin.BLE.Abstractions[0x7f9a6ab400] -> System.Runtime.Extensions[0x7f7b642980]: 3
Adv rec [Type CompleteLocalName; Data 49-44-31-30-37-20-48-52]
Adv rec [Type Appearance; Data 41-03]
Adv rec [Type Flags; Data 06]
Adv rec [Type UuidsComplete16Bit; Data 0A-F0]
Adv rec [Type ManufacturerSpecificData; Data 11-00-02-00-EA-7A-9B-5B-50-4E-0B]
[0:] Trying to reconnect
[0:] Guid Found
[0:] Scanning
Adapter: Stopping the scan for devices.
Adapter >= 21: Stopping the scan for devices.
```

But once I minimize the application. It doesn't "receive" or discover anything.

```
[0:] BLE Bluetooth is On
Thread started: <Thread Pool> #8
[0:] BLE IS SCANNING
Adapter >= 21: Starting a scan for devices.
06-30 15:38:49.147 D/Mono    (27664): [0x7f7863f440] worker starting
06-30 15:38:49.150 I/mono-stdout(27664): Adapter >= 21: Starting a scan for devices.
06-30 15:38:49.154 D/BluetoothAdapter(27664): STATE_ON
06-30 15:38:49.158 D/BluetoothLeScanner(27664): onClientRegistered() - status=0 clientIf=7
Adapter: Scan timeout has elapsed.
06-30 15:39:09.205 I/mono-stdout(27664): Adapter: Scan timeout has elapsed.
Adapter: Stopping the scan for devices.
Adapter >= 21: Stopping the scan for devices.
06-30 15:39:09.209 I/mono-stdout(27664): Adapter: Stopping the scan for devices.
06-30 15:39:09.210 I/mono-stdout(27664): Adapter >= 21: Stopping the scan for devices.
06-30 15:39:09.219 D/BluetoothAdapter(27664): STATE_ON
[0:] BLE Bluetooth is On
```

So my concern is, how to get the scanning to receive the device in the background(as in minimized/when the application is not in focus).

smstuebe, 2016-06-30:
There is an issue template ;) Please use it next time instead of deleting it. Please provide information about platform, plugin version, e.t.c.

LzJustin, 2016-06-30:
Right. Sorry.

Visual Studio 2015
Xamarin.Android
Plugin.BLE version beta4
Compiled using latest Platform (6.0 marshmallow)

LzJustin, 2016-07-01:
This is the sample of the Service demo we've tried. In focus/foreground of the application, it'll definitely reconnect. But not when it's minimized.

```
public class SimpleService : Service
    {
        private string _deviceID;
        private Plugin.BLE.Abstractions.Contracts.IAdapter _adapter;
        private IDevice currDevice;
        private IBluetoothLE _bleDevice;
        private System.Timers.Timer _timer;

        public override void OnCreate()
        {
            base.OnCreate();
            Console.WriteLine("OnCreate started");
            _bleDevice = CrossBluetoothLE.Current;
            _adapter = CrossBluetoothLE.Current.Adapter;
            _adapter.ScanTimeout = 20000;
	    _deviceID = "d8de624e-140f-4a22-8594-e2216b84a5f2"; //This is just a dummy GUID
	    _bleDevice.Adapter.DeviceDiscovered += reconnect; 
            _bleDevice.Adapter.DeviceConnected += Adapter_DeviceConnected;           
        }

        public override StartCommandResult OnStartCommand(Intent intent, StartCommandFlags flags, int startId)
        {
            Scanning(this, null);

            return StartCommandResult.NotSticky;
        }

        private async void Adapter_DeviceConnected(object sender, DeviceEventArgs e)
        {
            if (_adapter.IsScanning)
            {
                await _adapter.StopScanningForDevicesAsync();
            }

            currDevice = e.Device;
            System.Diagnostics.Debug.WriteLine("Connected");
        }

        public async void Scanning(object sender, EventArgs e)
        {
            await Task.Delay(5000); //We need the delay to have enough time to bring the app to background (out of focus/minimized)
            if (_bleDevice.IsAvailable)
            {
                if (_bleDevice.IsOn)
                {
                    System.Diagnostics.Debug.WriteLine("BLE Bluetooth is On");
                    if (!_bleDevice.Adapter.IsScanning)
                    {
                        System.Diagnostics.Debug.WriteLine("BLE IS SCANNING");
                        await _adapter.StartScanningForDevicesAsync();                        
                    }
                }
            }
        }

        public void reconnect(object sender, DeviceEventArgs e)
        {
            System.Diagnostics.Debug.WriteLine("Trying to reconnect");
            Guid id = new Guid();

            if (!Guid.TryParse(_deviceID, out id))
            {
                System.Diagnostics.Debug.WriteLine("Cannot find GUID");
                id = new Guid();
            }
            
            if (e.Device.Id == id)
            {
                System.Diagnostics.Debug.WriteLine("Guid Found");
                if (_adapter.IsScanning)
                {
                    System.Diagnostics.Debug.WriteLine("Scanning");
                    _adapter.StopScanningForDevicesAsync();
                }

                _adapter.ConnectToDeviceAync(e.Device);
            }

        }

        public override IBinder OnBind(Intent intent)
        {
            // This example isn't of a bound service, so we just return NULL.
            return null;
        }

        public override void OnDestroy()
        {
            base.OnDestroy();
            Log.Debug(TAG, "SimpleService destroyed at {0}.", DateTime.UtcNow);
        }
    }
}
```

LzJustin, 2016-07-12:
Any updates?

So far I've tried the 1.1 version for the "ConnectToPreviousKnownDevice" in order for it to work since the background scanning is not receiving.

smstuebe, 2016-07-12:
Not yet. We are focusing on 1.0. Do you have a minimal sample project on github? It would accelerate the process, when somebody is starting to investigate this issue.

hihdFu7y8B76nkEXR44BERcDp, 2017-01-06:
Has there beeen a change in this issue ?

hihdFu7y8B76nkEXR44BERcDp, 2017-01-06:
Solved it. I used the service object reference previously initalised from within a user thread.
I moved this initialization to function OnStartCommand() (within the service class), after which it worked.
Foreground/background - same performance.
My implementation uses the raw Android native API.

Kushal5, 2020-03-25:
^ Hi do you have some sample code for this