dotnet-bluetooth-le/dotnet-bluetooth-le #60: Support for LongRead

Dragon160, 2016-07-07:
For reading characteristics that are longer that the usual 20 bytes BLE offers a read option called (LongRead). It would be cool if your addin provides support for that.

> attclient_read_long command: Starts a procedure where the client first sends normal read request to the server, and if the server returns an attribute value with a length equal to the BLE MTU (22 bytes), then the client continues to send "read long" requests until rest of the attribute is read. This only applies if you are reading attributes that are longer than 22 bytes. It is often simpler to construct your GATT server such that there are no long attributes, for simplicity. Note that the BLE protocol still requires that data is packetized into max. 22-byte chunks, so using "read long" does not save transmission time

(https://bluegiga.zendesk.com/entries/25053373--REFERENCE-BLE-master-slave-GATT-client-server-and-data-RX-TX-basics)

I think there are two ways to support LongRead:

a) Always take LongRead since it only applies if the attribute is longer than usual
b) Check the size of the characteristics and than apply _Read_ or _LongRead_

smstuebe, 2016-07-07:
Option a) seems risky. Before we implement this we have to check wich versions/devices support this feature.
Option b) wich size do you refer?

Dragon160, 2016-07-07:
Hm theoretically we need to check if the MTU is smaller than the characteristics size.
This would be "automatically support for LongRead".

Another option would to introduce a new Command + native Implementation. Maybe this would be better since the client can choose between both reading options

smstuebe, 2016-07-07:
Btw: What's about LongWrite and LongNotify?

Dragon160, 2016-07-07:
Good point. 
Right now I am investigating how to use a ReadLong within Android/iOS - there is no API available for this. I hope that is not a build-by-our-own feature ;-)

smstuebe, 2016-07-08:
Bluetooth Spec (4.2) https://www.bluetooth.org/DocMan/handlers/DownloadDoc.ashx?doc_id=286439

- §4.8.3 Read Long Characteristic Values
- §4.9.4 Write Long Characteristic Values
- §4.12.2 Read Long Characteristic Descriptors
- §4.12.4 Write Long Characteristic Descriptors

I couldn't find something about "long notify".

midix, 2017-03-22:
I have been doing some investigation lately and getting somewhat desperate because there seem to be too few people knowing about how to leverage larger ATT MTU sizes (not confuse with new Bluetooth 4.2 data packet length extension feature, which is completely different thing and is supported only on very new devices) and, consequently, larger notifications/reads/writes. All tutorials I've seen just use 20 byte chunks and stop at that, while other articles (like this one https://punchthrough.com/blog/posts/maximizing-ble-throughput-part-2-use-larger-att-mtu )  demonstrate how they used larger MTU sizes and gained additional 15% of bandwidth, but unfortunately they give no code examples.

So, it all seems to be related to `requestMtu` and `onMtuChanged` on Android and `central.maximumUpdateValueLength` on iOS.

Quoting someone from StackOverflow:

> You can make the MTU bigger, as a request from the central using: requestMtu(...) API. The latter will cause a callback call onMtuChanged at the peripheral side which will inform it of the new MTU.

And another quote for similar mechanics on iOS:

>Peripheral side: When chunking your data, make sure your chunk size is no larger than `central.maximumUpdateValueLength`. On iOS7, between an iPad 3 and iPhone 5, I've typically seen 132 bytes. If you're sending to multiple centrals, use the least common value.

And one more useful quote from some Bluetooth hardware developers:

> You can read 64 bytes from your phone, but you cant enable the notifications and receive all 64 bytes. The notifications and the indications support only the current MTU of your peripheral. Even if you change the MTU in your advertising procedure your client has to accept it in order to transfer all the payload in a notification packet. After setting up the suggested MTU on the peripheral you have to invoke a GATT_EXC_MTU_CMD in order to tell to the central to follow your set MTU. And it is up to the central if it will comply with the peripheral's suggestion.

It is not related to iOS nor Android, but explains how it usually is organized at lower level.