dotnet-bluetooth-le/dotnet-bluetooth-le #61: Cannot communicate after connection lost and reconnection

enryb, 2016-07-08:
Hello,
I used your plugin to build an application.
All is working without problem, but I found a little problem and I can't understand if it is a plugin bug or a problem in my application:

If I write something in a characteristic and then I start notification to another characteristic all works well.
But if connection is lost (if I remove battery from my device) and then I reconnect (my device say that I correctly connected), than I can't write anything (my device don't recieve nothing) and I cannot start nofication.

## Steps to reproduce

1. Connect to a device
2. Write a value in a characteristic
3. Start notification to another characteristic
4. Disconnect device
5. Re connect to device
6. Write a value in a characteristic
7. Start notification to another caracteristic

## Expected behavior
Values should be written without problems and notification should starts.

## Actual behavior
No values are written and it is not possible to start a notification.

### Crashlog

[...]
GattCallback state: Disconnected
Disconnected by lost connection
DisconnectedPeripheral by lost signal: ***DEVICE_NAME***
07-08 17:21:24.535 D/BluetoothGatt(15023): onClientConnectionState() - status=0 clientIf=3 device=***DEVICE_ADDRESS***
07-08 17:21:24.535 D/BluetoothDevice(15023): mAddress: ***DEVICE_ADDRESS***
07-08 17:21:24.535 I/mono-stdout(15023): GattCallback state: Disconnected
07-08 17:21:24.536 D/BluetoothDevice(15023): mAddress: ***DEVICE_ADDRESS***
07-08 17:21:24.536 D/BluetoothDevice(15023): mAddress: ***DEVICE_ADDRESS***
07-08 17:21:24.536 I/mono-stdout(15023): Disconnected by lost connection
07-08 17:21:24.537 D/BluetoothDevice(15023): mAddress: ***DEVICE_ADDRESS***
07-08 17:21:24.538 D/BluetoothGatt(15023): close()
07-08 17:21:24.538 D/BluetoothGatt(15023): unregisterApp() - mClientIf=3
07-08 17:21:24.540 I/mono-stdout(15023): DisconnectedPeripheral by lost signal: ***DEVICE_NAME***
[...]
GattCallback state: Connected
Connected
07-08 17:21:30.861 I/mono-stdout(15023): GattCallback state: Connected
07-08 17:21:30.861 I/mono-stdout(15023): Connected
07-08 17:21:30.862 D/BluetoothDevice(15023): mAddress: ***DEVICE_ADDRESS***
07-08 17:21:30.862 D/BluetoothDevice(15023): mAddress: ***DEVICE_ADDRESS***
07-08 17:21:30.863 D/BluetoothDevice(15023): getName: name = ***DEVICE_NAME***
07-08 17:21:30.864 D/BluetoothDevice(15023): mAddress: ***DEVICE_ADDRESS***
07-08 17:21:30.864 D/BluetoothDevice(15023): mAddress: ***DEVICE_ADDRESS***
ConnectToDeviceAync Connected: 00000000-0000-0000-0000-***DEVICE_ADDRESS*** ***DEVICE_NAME***
07-08 17:21:30.865 I/mono-stdout(15023): ConnectToDeviceAync Connected: 00000000-0000-0000-0000-***DEVICE_ADDRESS*** ***DEVICE_NAME***
[...]
Characteristic.WriteAsync
07-08 17:21:34.068 I/mono-stdout(15023): Characteristic.WriteAsync
Write ***FIRST_CHARACTERISTIC***
07-08 17:21:34.101 I/mono-stdout(15023): Write ***FIRST_CHARACTERISTIC***
07-08 17:21:34.102 D/BluetoothGatt(15023): writeCharacteristic() - uuid: ***FIRST_CHARACTERISTIC***
Characteristic.StartUpdates
07-08 17:21:34.105 I/mono-stdout(15023): Characteristic.StartUpdates
07-08 17:21:34.106 D/BluetoothGatt(15023): setCharacteristicNotification() - uuid: ***SECOND_CHARACTERISTIC*** enable: true
07-08 17:21:34.203 D/Surface (15023): Surface::setBuffersDimensions(this=0xf45dad00,w=736,h=1280)
07-08 17:21:34.207 D/Surface (15023): Surface::setBuffersDimensions(this=0xf45dad00,w=736,h=1280)
07-08 17:21:34.211 D/Surface (15023): Surface::setBuffersDimensions(this=0xf45dad00,w=736,h=1280)
Descriptor set value: NOTIFY
07-08 17:21:34.215 I/mono-stdout(15023): Descriptor set value: NOTIFY
Characteristic.StartUpdates, successful: False
[...]

## Configuration

**Version:** 1.0.0 - beta 5

**Platform:** Android

**Device:** Lenovo Tab 2 a8-50f (Android 5.1 - Api 22) and LGE LG-X150 (Android 5.0 - Api 21)

Thanks for your attention,
Enryb

adolgov, 2016-07-08:
Upon re-connecting, did you scan for the services and characteristics, or were you attempting to interact with a service/characteristic object from the previous (initial) connection?

enryb, 2016-07-11:
Yes, I scan again for services and characteristics (I also check with breakpoints and the code is executed).
I scan directly to 
IDevice thisDev = adapter.ConnectedDevices[0];
(there is only one connected device).

The code I execute (and that works well the first time I use) is:

var service = await thisDev.GetServiceAsync(Guid.Parse("***ADDRESS***"));
            var characteristic = await service.GetCharacteristicAsync(Guid.Parse("***ADDRESS***"));

            await characteristic.WriteAsync(new byte[] { 0x01 });

            var toRead = await service.GetCharacteristicAsync(Guid.Parse("***ADDRESS***"));
            toRead.ValueUpdated += Aggiornamento;
            toRead.StartUpdates();

xabre, 2016-07-15:
Can you show me the code you use to re-connect the device?

Pispa339, 2016-08-18:
I'm having the same issue. Only on Android though, works fine on iOS.
```C#
private void OnDeviceConnectionLost(object sender, DeviceErrorEventArgs e)
		{
			_characteristic.StopUpdates();
			_adapter.ConnectToDeviceAsync(e.Device);
		}
```

```C#
private void OnDeviceConnected(object sender, DeviceEventArgs e)
		{
			BLEDevice = e.Device;
			StartService();
		}
```

```C#
private async Task startService()
		{
			_service = await BLEDevice.GetServiceAsync(Guid.Parse(serviceUUID));
			_characteristic = await _service.GetCharacteristicAsync(Guid.Parse(characteristicUUID));

			_characteristic.ValueUpdated += Characteristic_ValueUpdated;

			_characteristic.StartUpdates();
                       //After reconecting I get: "Characteristic.StartUpdates, successful: False"
		}
```

bravekit, 2016-09-08:
I have some issue on Android. In my case I can restore communication with BLE device after:
1) Connection lost
2) rescan devices
3) connect to discovered (same) BLE device 
But scanning after each connection lost bad idea, I will try to find sollution...

mariusbloemhof, 2017-02-22:
I have the exact same problem (Android only, works fine on IOS). It is very easy to reproduce. 

1. Connect to device.
2. Lose connection / disconnect
3. Start Scanning for devices
4. Device in question (1) is found. 
5. Try to connect to it.
6. Error as below:

02-22 17:24:07.079 D/BluetoothAdapter(24515): onScanResult() - Device=98:7B:F3:76:7E:CF RSSI=-81
[0:] mvx:Diagnostic:249.18 Adapter.LeScanCallback: Shadow Device
[0:] mvx:Diagnostic:249.19 Adv rec [Type Flags; Data 06]
[0:] mvx:Diagnostic:249.20 Adv rec [Type UuidsIncomple16Bit; Data 18-04-18-02-18-03]
[0:] mvx:Diagnostic:249.20 Adv rec [Type CompleteLocalName; Data 53-68-61-64-6F-77-20-44-65-76-69-63-65]
[0:] mvx:Diagnostic:249.22 Adapter.LeScanCallback: 
02-22 17:24:07.120 D/BluetoothAdapter(24515): onScanResult() - Device=14:99:E2:20:A1:72 RSSI=-94
[0:] mvx:Diagnostic:249.23 Adv rec [Type Flags; Data 1A]
[0:] mvx:Diagnostic:249.23 Adv rec [Type ManufacturerSpecificData; Data 4C-00-09-06-03-05-0A-64-6D-1F]
02-22 17:24:07.150 D/BluetoothGatt(24515): onClientRegistered() - status=0 clientIf=8
**02-22 17:24:07.150 E/BluetoothGatt(24515): Bad connection state: 0**
.
.
.
02-22 17:24:07.780 D/BluetoothGatt(24515): onClientConnectionState() - status=0 clientIf=8 device=98:7B:F3:76:7E:CF
02-22 17:24:07.780 W/BluetoothGatt(24515): Unhandled exception in callback
02-22 17:24:07.780 W/BluetoothGatt(24515): java.lang.NullPointerException
02-22 17:24:07.780 W/BluetoothGatt(24515): 	at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:172)
02-22 17:24:07.780 W/BluetoothGatt(24515): 	at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:71)
02-22 17:24:07.780 W/BluetoothGatt(24515): 	at android.os.Binder.execTransact(Binder.java:404)
02-22 17:24:07.780 W/BluetoothGatt(24515): 	at dalvik.system.NativeStart.run(Native Method)
02-22 17:24:08.351 D/BluetoothAdapter(24515): onScanResult() - Device=C9:43:AB:64:59:B5 RSSI=-69