dotnet-bluetooth-le/dotnet-bluetooth-le #64: ValueUpdated callback never fired

darcon77, 2016-07-25:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide a much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

1. Download the xamarin-bluetooth-le GitHub repo from master 

2. Replace all BLE plugin project references in BLE.Client and BLE.Client.Droid with NuGet packages from -pre (1.0.0 - beta 5)

3. Build and run Droid app. The CHaracteristic updates are no longer handled.

## Expected behavior
I have ran this app with solution project references to the BLE plugin and verified that the characteristics that I'm trying to listen to are updating.

## Actual behavior
After I referenced the plugin from NuGet, the updates stopped.

### Crashlog

If something causes an exception paste full stack trace + Exception here

## Configuration

**Version:** 1.0.0 -beta 5

**Platform:** Android

**Device:** Samsung Galaxy Tab E 8

ClausElmann, 2016-07-26:
Same here Android API 23
Works in IOS!

ClausElmann, 2016-07-27:
I resolved this by adding a 500 ms wait after myCharacteristic.StartUpdates();

xabre, 2016-07-27:
I'm trying to reproduce the described bug. I've added the **nuget package** to the sample app and deployed it in **release** mode. With my BLE test device I can't seem to reproduce it.
Do you have other test phones/devices? Does it also happen on those devices?

[image]

lucianomireles, 2016-08-01:
I had the same issue, on a Samsung SM-G900v 
Android 6.0 API 23.

However after trying **ClausElmann** Suggestion i had no improvement. So i dug a bit deeper and found out for some reason instead of using  the implementation from "Plugin.BLE.Android". It was trying to use just the standard plc Implementation "Plugin.BLE" which of course needs to be overloaded or else it does nothing.

To demonstrate this you can
1. Download the source code.
2. On the android project only use references from "Plugin.BLE" and  "Plugin.BLE.Abstractions".

How i fixed it
Make sure you function calls are using the correct libraries and not the generic PCL.

xabre, 2016-08-02:
I'm investigating it :) Thanks for the feedback

xabre, 2016-08-02:
Based on the size of the dll's linked to each platform, everything seems to be in order: 
- MonoAndroid:  **35KB**
- PCL: **5KB**
- Xamarin.iOS: **26KB**

[image]

Qenadin, 2016-08-02:
Ah, well when I install the package using visual studios and nuget, it looks close to what you have.
[image]
 HOWEVER my object browser doesn't even see Plugin.BLE (probably because there's no references for it to browse). So when i actually build the project, you can see in my build folder the Plugin.BLE.Android is not built with my project. 
[image]

Regardless its simple enough to download the source and just make sure you reference Plugin.BLE.Android. I'm not sure what is up with that nuget package tho.

xabre, 2016-08-02:
@Qenadin The assembly file name is always "Plugin.BLE.dll", the contents are different for each platform. For android the assembly contains the "Plugin.BLE.Android" namespace ;). 33KB relative to 35KB might be due to win vs os-x counting bytes :P. 

This has to work by using the nuget package.... Will try to investigate further. Do you have a sample solution which you can share?

darcon77, 2016-08-08:
I can confirm that after updating to v1.0.0 stable (not pre) the issue is gone.

xabre, 2016-08-11:
Great :)

closed 2016-08-11 by xabre

sumitgouthaman, 2016-08-18:
I am running into this issue on v1.0.0 when I try to subscribe to multiple characteristics at the same time. Any pointers?

darcon77, 2016-08-18:
add a delay between calls to StartUpdate. 1 second was enough for me

Sent from my Verizon, Samsung Galaxy smartphone<div>
</div><div>
</div><div>-------- Original message --------</div><div>From: Sumit Gouthaman <notifications@github.com> </div><div>Date: 8/17/16  10:09 PM  (GMT-05:00) </div><div>To: xabre/xamarin-bluetooth-le <xamarin-bluetooth-le@noreply.github.com> </div><div>Cc: darcon77 <darcon77@hotmail.com>, Author <author@noreply.github.com> </div><div>Subject: Re: [xabre/xamarin-bluetooth-le] ValueUpdated callback never fired (#64) </div><div>
</div>
I am running into this issue on v1.0.0 when I try to subscribe to multiple characteristics at the same time. Any pointers?

sumitgouthaman, 2016-08-18:
@darcon77 : That worked thanks. Does this mean there is still some aspect of this bug that isn't resolved?