dotnet-bluetooth-le/dotnet-bluetooth-le #65: Compile time error using Plugin.BLE  beta 5

leelatoo, 2016-07-28:
Hi,

We are currently using Plugin.BLE bet3 in our project, which works fine. When we update the Plugin.BLE nuget package to beta 5, we are not able to succesfully compile the project. It gives "MT2002" error.
Is it because the xamarin environment? if so, what is the version of Xamarin.iOS or Xamarin required?

Can you please help me solve the mentioned problem?

Thanks in advance

Leela

xabre, 2016-07-29:
We used the latest stable Xamarin version to compile it. Did you update Xamarin to the latest stable release? Perhaps you can try cloning the repository and build the solution.

smstuebe, 2016-08-12:
@leelatoo do you have this issue with 1.0.0 stable, too?

xabre, 2016-09-22:
No activity since Jul 28. Will consider this closed

closed 2016-09-22 by xabre