dotnet-bluetooth-le/dotnet-bluetooth-le #67: Plugin.BLE beta 5, GattCallback error: 133 from ConnectToDeviceAsync(IDevice)

nbodinripert, 2016-08-01:
Hi,

I'm using Plugin.BLE beta 5 for my project but I get an exception when I try to connect my phone to a BLE device.

## Steps to reproduce

1. I scan for all of devices I can in a page DevicesListPage.

2. I choose the device I want to connect to and I passed it through a new Page, ServicesListPage, with that constructor : ServicesListPage(IDevice device). This page has a field, _connectedDevice. I set it in the constructor like this:

`_connectedDevice = device`

3. The page ServicesListPage has a button "Connect" to connecting my phone to the device _connectedDevice. When I click on the button the method Connection() is calling:

```
async void Connection(object sender, EventArgs e)
{
	if (_connectedDevice != null && _connectedDevice.State == Plugin.BLE.Abstractions.DeviceState.Disconnected)
	{
		try
		{
			await App.Adapter.ConnectToDeviceAync(_connectedDevice);
		}
		catch (DeviceConnectionException e){
			System.Diagnostics.Debug.WriteLine("EXCEPTION\nDevice : " + e.DeviceId);
			System.Diagnostics.Debug.WriteLine("Message : " + e.Message + "\nStackTrace : " + e.StackTrace);
		}
	}
}
```

## Expected behavior
Connection should succeed.

## Actual behavior
I got a DeviceConnectionException with this message

> GattCallback error: 133

### Crashlog

> EXCEPTION
> Device : 00000000-0000-0000-0000-d03972ba4265
> Message : GattCallback error: 133
> StackTrace :   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in /Users/builder/data/lanes/3415/7db2aac3/source/mono/external/referencesource/mscorlib/system/runtime/exceptionservices/exceptionservicescommon.cs:143 
>   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (System.Threading.Tasks.Task task) [0x00047] in /Users/builder/data/lanes/3415/7db2aac3/source/mono/external/referencesource/mscorlib/system/runtime/compilerservices/TaskAwaiter.cs:187 
>   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Threading.Tasks.Task task) [0x0002e] in /Users/builder/data/lanes/3415/7db2aac3/source/mono/external/referencesource/mscorlib/system/runtime/compilerservices/TaskAwaiter.cs:156 
>   at System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd (System.Threading.Tasks.Task task) [0x0000b] in /Users/builder/data/lanes/3415/7db2aac3/source/mono/external/referencesource/mscorlib/system/runtime/compilerservices/TaskAwaiter.cs:128 
>   at System.Runtime.CompilerServices.TaskAwaiter`1[TResult].GetResult () [0x00000] in /Users/builder/data/lanes/3415/7db2aac3/source/mono/external/referencesource/mscorlib/system/runtime/compilerservices/TaskAwaiter.cs:357 
>   at Plugin.BLE.Abstractions.Utils.TaskBuilder+<FromEvent>d__1`3[TReturn,TEventHandler,TRejectHandler].MoveNext () [0x000c0] in C:\Users\sead\Documents\GitHub\xamarin-bluetooth-le\Source\Plugin.BLE.Abstractions\Utils\TaskBuilder.cs:51 
> --- End of stack trace from previous location where exception was thrown ---
>   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in /Users/builder/data/lanes/3415/7db2aac3/source/mono/external/referencesource/mscorlib/system/runtime/exceptionservices/exceptionservicescommon.cs:143 
>   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (System.Threading.Tasks.Task task) [0x00047] in /Users/builder/data/lanes/3415/7db2aac3/source/mono/external/referencesource/mscorlib/system/runtime/compilerservices/TaskAwaiter.cs:187 
>   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Threading.Tasks.Task task) [0x0002e] in /Users/builder/data/lanes/3415/7db2aac3/source/mono/external/referencesource/mscorlib/system/runtime/compilerservices/TaskAwaiter.cs:156 
>   at System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd (System.Threading.Tasks.Task task) [0x0000b] in /Users/builder/data/lanes/3415/7db2aac3/source/mono/external/referencesource/mscorlib/system/runtime/compilerservices/TaskAwaiter.cs:128 
>   at System.Runtime.CompilerServices.TaskAwaiter.GetResult () [0x00000] in /Users/builder/data/lanes/3415/7db2aac3/source/mono/external/referencesource/mscorlib/system/runtime/compilerservices/TaskAwaiter.cs:113 
>   at BLETest.ServicesListPage+<ConnectionToDevice>c__async0.MoveNext () [0x0005a] in /Users/nbodinripert/Projects/BLETest/BLETest/Views/Pages/ServicesListPage.xaml.cs:105 

## Configuration

**Version:** 
Xamarin Studio 6.0
Xamarin.Forms : 2.0.1.6495
Plugin.BLE 1.0.0-beta5

**Platform:** Xamarin Forms (no specific code in iOS or Android sides) but I tested my app only on my Android phone.

**Device:** LGE Nexus 5X (API 23)

lucianomireles, 2016-08-01:
Your going to hate hearing this, but almost always when i receive this error its due to my device's Bluetooth module being incorrect not the code. Check your flow control on your module, make sure Rts and Cts is set properly.

nbodinripert, 2016-08-02:
Thanks a lot for your quick answer. I tried with an iPhone and it's worked.

Qenadin, 2016-08-02:
No problem, glad you got it functioning. Actually took  some time to read up on the LGE Nexus, and it seems it has alot of issues with Bluetooth in general. Perhaps trying a different type of android device? A Samsung may be better.

nbodinripert, 2016-08-02:
I had the same exception with a Samsung Galaxy S6 Edge ;)

Qenadin, 2016-08-02:
Are you calling it from the main ui thread?

nbodinripert, 2016-08-02:
I'm not familiar with the notion of ui thread sorry, but when I click on my "Connect" button I call directly the Connection() method, I don't write: 

```
Device.BeginInvokeOnMainThread (async () => {
  await App.Adapter.ConnectToDeviceAync(_connectedDevice);
});
```

xabre, 2016-08-02:
Did you try to connect to your BLE peripherial device using the sample app?

To run the sample app:
- clone repo & deploy the android csproj BLE.Client.Droid

nbodinripert, 2016-08-04:
No I did not. I will try tomorrow.

allbdeveloper, 2016-09-03:
http://stackoverflow.com/questions/25330938/android-bluetoothgatt-status-133-register-callback

I found this question.
Is this similar issue? 
Thanks always.

allbdeveloper, 2016-09-03:
I got same exception sometimes on GS 4.

rcbolt3, 2017-10-24:
I'm getting the same exception on my Samsung Galaxy S8+ using both my app and your sample app. However, everything runs smoothly on my Samsung tablet. Any ideas?

XamHans, 2018-08-14:
Any News to this ? Receiving the same error.
Can I connect to IBeacon despite it is password protected when I have the UID ?

SamuelLioult, 2018-11-13:
Hi,
I had the same issue. I resolved it specifying the transport type to the ConnectToDeviceAsync method like this:

```c#
var parameters = new ConnectParameters(forceBleTransport: true);
await _adapter.ConnectToDeviceAsync(deviceToConnect, parameters, cancellationToken.Token);
```
Hope it helps.

closed 2019-08-05 by xabre