dotnet-bluetooth-le/dotnet-bluetooth-le #68: Can't build BLE.sln on Windows 10/Visual Studio 2015 Community Ed.

gaminn, 2016-08-04:
Hi,
it is impossible to build the solution on Windows 10/Visual Studio 2015 Community Edition/latest Xamarin. The error is:

Error		No resource found that matches the given name: attr 'colorAccent'.	BLE.Client.Droid	D:\Projects\Test\xamarin-bluetooth-le\Source\BLE.Client\BLE.Client.Droid\Resources\values\style.xml	2	
Error		No resource found that matches the given name: attr 'colorPrimaryDark'.	BLE.Client.Droid	D:\Projects\Test\xamarin-bluetooth-le\Source\BLE.Client\BLE.Client.Droid\Resources\values\style.xml	2	
Error		No resource found that matches the given name: attr 'windowActionModeOverlay'.	BLE.Client.Droid	D:\Projects\Test\xamarin-bluetooth-le\Source\BLE.Client\BLE.Client.Droid\Resources\values\style.xml	2	
Error		Error retrieving parent for item: No resource found that matches the given name 'Theme.AppCompat.Light.NoActionBar'.	BLE.Client.Droid	D:\Projects\Test\xamarin-bluetooth-le\Source\BLE.Client\BLE.Client.Droid\Resources\values\style.xml	2	
Error		No resource found that matches the given name: attr 'colorPrimary'.	BLE.Client.Droid	D:\Projects\Test\xamarin-bluetooth-le\Source\BLE.Client\BLE.Client.Droid\Resources\values\style.xml	2

smstuebe, 2016-08-04:
Have you updated your android SDK tools, the support library and the platform?

gaminn, 2016-08-04:
I installed all last week from internet.

smstuebe, 2016-08-05:
Have you double checked by opening the Android SDK manager?

gaminn, 2016-08-08:
I have:
Android SDK Tools rev. 25.1.7
Android 7.0 (API 24) SDK Platform 24, rev. 2
I don't know what the support library is. I don't see it in Android SDK Manager.

smstuebe, 2016-08-08:
AndroidUseLatestPlatformSdk is currently set to false and the compile sdk to 6.0
Can you try to install Android 6.0 (API 23) or change the "compile using android Version" setting to "7.0", "use latest", respectively in the project settings of the android project?

gaminn, 2016-08-09:
I changed target SDK to several lower versions and it doesn't help.

smstuebe, 2016-08-09:
Support library is in "Extras" group.

[image]

a) Do you have this?
b) Have you updated some nuget packages of the sample project?

I can't really help you more :< The automated build on bitrise is running and it is compiling on our machines

gaminn, 2016-08-10:
Yes, I have the latest Android Support Library. I haven't seen it as it was hidden under obsolete checkbox.

I tried to update all nuget packages, but during update I get the same errors a when building the solution.

OK, consider this issue unresolved.

smstuebe, 2016-08-10:
@xabre do you have any idea?

xabre, 2016-08-10:
I'll take a look. I'dont have API level 24 installed on my machines. Also check this: [https://releases.xamarin.com/technical-bulletin-android-sdk-build-tools-24/](https://releases.xamarin.com/technical-bulletin-android-sdk-build-tools-24/). 

@gaminn Just one quick question, if you create a new empty xamarin forms app in VS can you build the android app, just like that without any changes? 

Also make sure not to update the nuget packages in the solution for xamarin android support (even if vs usggests it). Xamarin forms has a fixed dependecy on i think (=23.3.0).

gaminn, 2016-08-10:
Yes, no problem in creating blank Xamarin.Forms solution and then building it.

xabre, 2016-08-10:
I've just installed API level 24 together with the latest sdk build tools (24.0.1). I had to update JDK to 1.8 (Java8) but after that everything builds smoothly. 

@gaminn My current suggestion is :

- clone a fresh copy (or git reset --hard and git clean -f -d- x) 
- checkout the 1.1.0 branch (we just updated all the nuget packages in this branch to the latest supported version)
- restore nuget pacakges (right clicok on solutin -> Restore)
- check that the Xam.AndroidSupport.V7 library (see screenshot1) is installed correctly
- check that xamarin correctly downloaded the Android Java repository for the support lib (see screenshot 2). 
- rebuild

The errors you are experiencing is most likely cause by something related to this specific support library Theme.AppCompat.* come from here.

**Screen1: VS** 
[image]

**Screen2: Xamarin AppData**
[image]

xabre, 2016-08-15:
@gaminn any news in this one?

gaminn, 2016-08-15:
> check that xamarin correctly downloaded the Android Java repository for the support lib (see screenshot 2). 

My AppData\Local\Xamarin\Xamarin.Android.Support.v7.AppCompat\23.3.0.0 folder is empty. How to fix it?

smstuebe, 2016-08-15:
Delete and rebuild

xabre, 2016-08-16:
`git clean -f -d- x`
also delete the empty folders in AppData

gaminn, 2016-08-17:
I downloaded android_m2repository_r29.zip (as Visual Studio instructed me to download it and extract it in 23.3.0.0 folder) and extracted in AppData\Local\Xamarin\Xamarin.Android.Support.v7.AppCompat.

Still the same errors.

I would like to thank you for your support but it seems we are not able to resolve this issue. Please close it, I'm not longer interested in buiding the solution.

closed 2016-08-17 by smstuebe