dotnet-bluetooth-le/dotnet-bluetooth-le #69: Plain Vanila Plugin crash when scanning with GUID.

alberk8, 2016-08-09:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide a much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

1.
//This statement crash if I have the GUID, without any parameter it works but it would return 
//every BLE device in the vicinity.
_bleAdapter.StartScanningForDevicesAsync(new[] { Guid.Parse("ffe0ecd2-3d16-4f8d-90de-e89e7fc396a5") });
2.

3.

## Expected behavior
Tell us what should happen
The Plugin Should be able to scan for the specific device by GUID

## Actual behavior
Tell us what happens instead
The App Crash

### Crashlog

If something causes an exception paste full stack trace + Exception here

[image]
[image]

## Configuration

**Version:**  
Plugin BLE 1.0.0 Release
Xamarin 4.1.2.18
Xamarin.Android 6.1.2.21
Visual Studio 1514.0.25421.03 Update 3

**Platform:** iOS/Android/ ...
Android 6.0.1 

**Device:** Xiaomi Redmi 3S..

smstuebe, 2016-08-09:
In your crash log I see `ConnectToDeviceAsync` are you sure, that it happens on `StartScanningForDevicesAsync`? Do you have more code or a minimal example?

alberk8, 2016-08-10:
Hi,

I did some more debugging on my code and it is true that the ConnectToDeviceAsync was called.  My mistake was not to have checked ` _adaptor.DiscoveredDevices.Count > 0` first.  When calling `ConnectToDeviceAsync( _device )` where the _device is null will cause the application to crash.

Is it possible to change the behavior of `ConnectToDeviceAsync`  to throw an Exception instead of blowing up?.

Thanks.

smstuebe, 2016-08-10:
ye, we can throw an `ArgumentNullException` on here.

closed 2016-08-10 by smstuebe