dotnet-bluetooth-le/dotnet-bluetooth-le #70: Sample project for Xamarin.Forms without MvvmCross

gaminn, 2016-08-09:
Hi,
is there a sample project for Xamarin.Forms without MvvmCross?

I'm trying to run BLE on Android, but I always get System.NotImplementedExcception: This functionality is not implemented in the portable version of this assembly.

I installed Plugin.BLE nuget package in Portable project and also in Droid project in my Xamarin.Forms solution. I get the exception when I reference Plugin.BLE and call CrossBluetoothLE.Current in Droid project (in MainActivity.cs) and I get also the same exception when I reference Plugin.BLE and call CrossBluetoothLE.Current in portable part of my solution (in App.xaml.cs).

I obviously don't know how to use cross platform code in my solution, so please help me.

alberk8, 2016-08-09:
Hi 
If you do not mind a very sample vanilla code.   I did not implement the Listview for user to select, just grep the first Device found and connect to it.  You can improve on that.  Thanks.

```
using System;
using System.Linq;
using Android.App;
using Android.Content;
using Android.Runtime;
using Android.Views;
using Android.Widget;
using Android.OS;
using Plugin.BLE;
using Plugin.BLE.Abstractions.Contracts;
using Plugin.BLE.Android;
using Plugin.BLE.BroadcastReceivers;
using Plugin.BLE.Abstractions.EventArgs;
using Plugin.BLE.Abstractions.Exceptions;
using Plugin.BLE.Abstractions.Extensions;

namespace BluetoothMonitor
{
    [Activity(Label = "BluetoothMonitor", MainLauncher = true, Icon = "@drawable/icon")]
    public class MainActivity : Activity
    {
    
        private Plugin.BLE.Abstractions.Contracts.IAdapter _bleAdapter;
     
        private IDevice currDevice;
        private IBluetoothLE _bluetooth;
     
        private Button scanButton;
        private Button connButton;
        private Button stopButton;
        private EditText edText;
        private DateTime datetime;

        protected override void OnCreate(Bundle bundle)
        {
            base.OnCreate(bundle);

            _bluetooth = CrossBluetoothLE.Current;
            _bleAdapter = CrossBluetoothLE.Current.Adapter;
            _bleAdapter.ScanTimeout = 10000;
           

            // Set our view from the "main" layout resource
            SetContentView(Resource.Layout.Main);

            // Get our button from the layout resource,
            // and attach an event to it
            stopButton = FindViewById<Button>(Resource.Id.stopbutton);
            scanButton = FindViewById<Button>(Resource.Id.scanbutton);
            connButton = FindViewById<Button>(Resource.Id.connblebtn);
            edText = FindViewById<EditText>(Resource.Id.edittext);

          

            //button.Click += delegate { button.Text = string.Format("{0} clicks!", count++); };
            stopButton.Click += delegate { this.Finish(); };
            scanButton.Click += bleScan; 
            connButton.Click += bleconnect;

            _bleAdapter.DeviceDiscovered += bleDiscovered;

        }

        public void bleScan(object sender, EventArgs e)
        {
            _bleAdapter.StartScanningForDevicesAsync();
        }

        public async void bleDiscovered(object sender, DeviceEventArgs e)
        {
            currDevice = e.Device;
            //Grep the first device only.  
            if (_bleAdapter.IsScanning)
                await _bleAdapter.StopScanningForDevicesAsync();

            
        }

        public async void bleconnect(object sender, EventArgs e)
        {
            string errorstr ="" ;
            TimeSpan scantime; 
            if(_bluetooth.IsAvailable)
                if(_bluetooth.IsOn)
                {
                    datetime = DateTime.Now;
                    
                    try
                    {
                        edText.Text = "Start Scanning...";
                        
                        if (_bleAdapter.ConnectedDevices.Count > 0)
                        {
                            Toast.MakeText(this, "Disconnecting BLE Device" , ToastLength.Short).Show();
                            await _bleAdapter.DisconnectDeviceAsync(currDevice);
                            
                        }
                        await _bleAdapter.StartScanningForDevicesAsync();
                        scantime = DateTime.Now - datetime;
                    }
                    catch(DeviceDiscoverException er )
                    {
                        errorstr = er.Message;
                    }

                    Toast.MakeText(this, "After Try Catch ", ToastLength.Short).Show();

                     scantime = DateTime.Now - datetime;
                    if (errorstr =="")
                        edText.Text = "Scan Time in ms :" + scantime.TotalMilliseconds.ToString() + "\n";
                    else
                        edText.Text = errorstr + " :" + scantime.TotalMilliseconds.ToString() + "\n";

                    datetime = DateTime.Now;
                    errorstr = "";

                    if (_bleAdapter.DiscoveredDevices.Count > 0)
                    {
                        try
                        {
                            await _bleAdapter.ConnectToDeviceAsync(currDevice, false);
                        }
                        catch (DeviceConnectionException er)
                        {
                            errorstr = er.Message;
                        }
                        var conntime = DateTime.Now - datetime;
                        if (errorstr == "")
                            edText.Text += "Connection Time in ms :" + conntime.TotalMilliseconds.ToString();
                        else
                            edText.Text += errorstr + " :" + conntime.TotalMilliseconds.ToString();
                    }
                    else
                        edText.Text += "No Device Found";

                }
        }
    }
}
```

gaminn, 2016-08-09:
Thanks for the code but this is what I already tried. I get the not implemented exception on CrossBluetoothLE.Current call in MainActivity.cs in Droid project of my Xamarin.Forms cross platform solution.

alberk8, 2016-08-09:
Are you running in a physical hardware or emulator?

gaminn, 2016-08-09:
Physical Hardware - LG G3 (Android 5.0 API 21).

alberk8, 2016-08-09:
To clarify, Are you saying that you are getting the exception when running my sample code?

gaminn, 2016-08-09:
Almost. I guess your code is for pure Android project, not for Xamarin.Forms cross platform app. I get the exception when running this code in my Xamarin.Forms cross platform app on Android device.

`// MainActivity.cs
using System;

using Android.App;
using Android.Content.PM;
using Android.Runtime;
using Android.Views;
using Android.Widget;
using Android.OS;

using Plugin.BLE;
using Plugin.BLE.Abstractions.Contracts;
using Plugin.BLE.Android;
using Plugin.BLE.BroadcastReceivers;
using Plugin.BLE.Abstractions.EventArgs;
using Plugin.BLE.Abstractions.Exceptions;
using Plugin.BLE.Abstractions.Extensions;

namespace BlankAppXamlXamarinForms.Droid
{
    [Activity(Label = "BlankAppXamlXamarinForms", Icon = "@drawable/icon", Theme = "@style/MainTheme", MainLauncher = true, ConfigurationChanges = ConfigChanges.ScreenSize | ConfigChanges.Orientation)]
    public class MainActivity : global::Xamarin.Forms.Platform.Android.FormsAppCompatActivity
    {
        IBluetoothLE ble;

        protected override void OnCreate(Bundle bundle)
        {
            TabLayoutResource = Resource.Layout.Tabbar;
            ToolbarResource = Resource.Layout.Toolbar;

            base.OnCreate(bundle);

            global::Xamarin.Forms.Forms.Init(this, bundle);

            ble = CrossBluetoothLE.Current; // System.NotImplementedExcception

            LoadApplication(new App());
        }
    }
}`

alberk8, 2016-08-09:
I am sorry as i am still very much not familiar with Xamarin.Forms and MVVM.  A bit too abstract for me at the moment.

gaminn, 2016-08-09:
No problem. Thanks for your time.

gaminn, 2016-08-10:
Is there anyone who can help me resolving System.NotImplementedException?

smstuebe, 2016-08-10:
@gaminn more info? o0

gaminn, 2016-08-10:
I created blank Xamarin.Forms solution in VS 2015. Then, I tried to use this BLE plugin. However, I always get System.NotImplementedException on CrossBluetoothLE.Current call no matter where I call it (in portable project - App.xaml.cs - or in Droid project - MainActivity.cs). I installed the BLE plugin using Install-Package Plugin.BLE in portable project and also in Droid project.

smstuebe, 2016-08-10:
@gaminn I've done the same, right now and it is working without any additional actions. Is your nuget up to date? `Tools -> Extensions and Updates` -> Updates -> Visual Studio Gallery`

gaminn, 2016-08-10:
It seems that Clean Solution command resolved my issue. I don't have a clue why...

closed 2016-08-10 by smstuebe

skrichten, 2016-08-22:
I am having this issue right now.  I have tried with both the latest stable release and the pre-release nuget packages.  I think I only need to install the package in my portable project, but I also tried installing in my .Droid project with same results.  I am calling CrossBluetoothLE.Current in my App.css (PCL) just for quick testing purposes.  Testing on HTC ONE M8 Android 4.4.  Any ideas?

smstuebe, 2016-08-22:
@skrichten how have you installed it to your .Droid project?  We tested it several times because of this discussion and it worked every time.

skrichten, 2016-08-22:
I just used the "manage nuget packages" feature in Visual studio.  What is the correct way to go about it?  Thanks!

skrichten, 2016-08-22:
So your hint that I may not have installed properly prompted me to try a different way.  So I uninstalled and tried again... installed from the solution instead of the individual projects.  I'm not seeing the error any more.