dotnet-bluetooth-le/dotnet-bluetooth-le #73: ConnectToKnownDeviceAsync Crash When Bluetooth is not in Range

alberk8, 2016-08-11:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide a much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

1.
Move Bluetooth device out of Range and run the APP to connect to it.
2.
await _bleAdapter.ConnectToKnownDeviceAsync(Guid.Parse("00000000-0000-0000-0000-f6bfd73398f7"));
3.

## Expected behavior
Tell us what should happen
The ConnectToKnownDeviceAsync Call should return an Exception with out crashing. 

## Actual behavior
Tell us what happens instead
The Call returns an Exception and then Crash the App.

### Crashlog
08-11 16:45:13.241 D/Mono    ( 7274): DllImport searching in: '__Internal' ('(null)').
08-11 16:45:13.241 D/Mono    ( 7274): Searching for 'java_interop_jnienv_call_boolean_method_a'.
08-11 16:45:13.241 D/Mono    ( 7274): Probing 'java_interop_jnienv_call_boolean_method_a'.
08-11 16:45:13.241 D/Mono    ( 7274): Found as 'java_interop_jnienv_call_boolean_method_a'.
08-11 16:45:13.278 D/Mono    ( 7274): DllImport searching in: '__Internal' ('(null)').
08-11 16:45:13.278 D/Mono    ( 7274): Searching for 'java_interop_jnienv_call_int_method_a'.
08-11 16:45:13.278 D/Mono    ( 7274): Probing 'java_interop_jnienv_call_int_method_a'.
08-11 16:45:13.278 D/Mono    ( 7274): Found as 'java_interop_jnienv_call_int_method_a'.
08-11 16:45:13.284 D/Mono    ( 7274): Assembly Ref addref Plugin.BLE.Abstractions[0x555c27e800] -> System.Threading.Tasks[0x555c291d00]: 2
08-11 16:45:13.284 D/Mono    ( 7274): Assembly Ref addref System.Threading.Tasks[0x555c291d00] -> mscorlib[0x555c2b6830]: 18
08-11 16:45:13.296 D/Mono    ( 7274): Assembly Ref addref Plugin.BLE.Abstractions[0x555c27e800] -> System.Collections[0x555c283920]: 3
08-11 16:45:13.310 D/Mono    ( 7274): DllImport searching in: '__Internal' ('(null)').
08-11 16:45:13.310 D/Mono    ( 7274): Searching for 'java_interop_jnienv_get_static_int_field'.
08-11 16:45:13.310 D/Mono    ( 7274): Probing 'java_interop_jnienv_get_static_int_field'.
08-11 16:45:13.311 D/Mono    ( 7274): Found as 'java_interop_jnienv_get_static_int_field'.
08-11 16:45:13.368 V/BoostFramework( 7274): mAcquireFunc method = public int com.qualcomm.qti.Performance.perfLockAcquire(int,int[])
08-11 16:45:13.368 V/BoostFramework( 7274): mReleaseFunc method = public int com.qualcomm.qti.Performance.perfLockRelease()
08-11 16:45:13.368 V/BoostFramework( 7274): mAcquireTouchFunc method = public int com.qualcomm.qti.Performance.perfLockAcquireTouch(android.view.MotionEvent,android.util.DisplayMetrics,int,int[])
08-11 16:45:13.368 V/BoostFramework( 7274): mIOPStart method = public int com.qualcomm.qti.Performance.perfIOPrefetchStart(int,java.lang.String)
08-11 16:45:13.368 V/BoostFramework( 7274): mIOPStop method = public int com.qualcomm.qti.Performance.perfIOPrefetchStop()
08-11 16:45:13.373 V/BoostFramework( 7274): BoostFramework() : mPerf = com.qualcomm.qti.Performance@1cb8f62
08-11 16:45:13.373 V/BoostFramework( 7274): BoostFramework() : mPerf = com.qualcomm.qti.Performance@12818f3
08-11 16:45:13.423 D/Mono    ( 7274): DllImport searching in: '__Internal' ('(null)').
08-11 16:45:13.423 D/Mono    ( 7274): Searching for 'java_interop_jnienv_call_nonvirtual_object_method_a'.
08-11 16:45:13.423 D/Mono    ( 7274): Probing 'java_interop_jnienv_call_nonvirtual_object_method_a'.
08-11 16:45:13.423 D/Mono    ( 7274): Found as 'java_interop_jnienv_call_nonvirtual_object_method_a'.
08-11 16:45:13.487 D/Mono    ( 7274): DllImport searching in: '__Internal' ('(null)').
08-11 16:45:13.487 D/Mono    ( 7274): Searching for 'java_interop_jnienv_call_void_method_a'.
08-11 16:45:13.487 D/Mono    ( 7274): Probing 'java_interop_jnienv_call_void_method_a'.
08-11 16:45:13.487 D/Mono    ( 7274): Found as 'java_interop_jnienv_call_void_method_a'.
08-11 16:45:13.492 D/Mono    ( 7274): Assembly Ref addref Plugin.BLE.Abstractions[0x555c27e800] -> System.Threading[0x555c280f50]: 3
08-11 16:45:13.512 D/OpenGLRenderer( 7274): Use EGL_SWAP_BEHAVIOR_PRESERVED: true
08-11 16:45:13.520 D/ActivityThreadInjector( 7274): clearCachedDrawables.
08-11 16:45:13.577 I/Adreno  ( 7274): QUALCOMM build                   : 3ab7efa, Iaaff4ce0ae
08-11 16:45:13.577 I/Adreno  ( 7274): Build Date                       : 07/21/16
08-11 16:45:13.577 I/Adreno  ( 7274): OpenGL ES Shader Compiler Version: XE031.07.00.01
08-11 16:45:13.577 I/Adreno  ( 7274): Local Branch                     : 
08-11 16:45:13.577 I/Adreno  ( 7274): Remote Branch                    : refs/tags/AU_LINUX_ANDROID_LA.UM.5.1_RB1.06.00.01.192.038
08-11 16:45:13.577 I/Adreno  ( 7274): Remote Branch                    : NONE
08-11 16:45:13.577 I/Adreno  ( 7274): Reconstruct Branch               : NOTHING
08-11 16:45:13.586 I/OpenGLRenderer( 7274): Initialized EGL, version 1.4
08-11 16:45:17.502 D/Mono    ( 7274): DllImport attempting to load: '__Internal'.
08-11 16:45:17.502 D/Mono    ( 7274): DllImport loaded library '(null)'.
08-11 16:45:17.503 D/Mono    ( 7274): DllImport searching in: '__Internal' ('(null)').
08-11 16:45:17.503 D/Mono    ( 7274): Searching for 'monodroid_get_system_property'.
08-11 16:45:17.503 D/Mono    ( 7274): Probing 'monodroid_get_system_property'.
08-11 16:45:17.503 D/Mono    ( 7274): Found as 'monodroid_get_system_property'.
08-11 16:45:17.503 D/Mono    ( 7274): DllImport searching in: '__Internal' ('(null)').
[... 40 lines cut ...]
08-11 16:45:48.008 D/Mono    ( 7274): DllImport searching in: '__Internal' ('(null)').
08-11 16:45:48.009 D/Mono    ( 7274): Searching for 'java_interop_jnienv_call_nonvirtual_boolean_method_a'.
08-11 16:45:48.009 D/Mono    ( 7274): Probing 'java_interop_jnienv_call_nonvirtual_boolean_method_a'.
08-11 16:45:48.009 D/Mono    ( 7274): Found as 'java_interop_jnienv_call_nonvirtual_boolean_method_a'.
Disconnected by user
08-11 16:45:48.012 I/mono-stdout( 7274): Disconnected by user
08-11 16:45:48.024 D/BluetoothGatt( 7274): close()
08-11 16:45:48.024 D/BluetoothGatt( 7274): unregisterApp() - mClientIf=5
08-11 16:45:48.026 I/mono-stdout( 7274): Error while connecting ''. Not raising disconnect event.
Error while connecting ''. Not raising disconnect event.
Unhandled Exception:

Plugin.BLE.Abstractions.Exceptions.DeviceConnectionException: GattCallback error: 133

## Configuration

**Version:** 1.1 (download source dated 11-Aug-2016)

**Platform:** Android 6.0.1
Visual Studio 2015 V14.0.25421.03 Update 3
Xamarin 4.1.2.18
Xamarin.Android 6.1.2.21
Android API 23

**Device:** Xiaomi Redmi 3S

xabre, 2016-08-11:
Well, this is more or less the intended behaviour. 

For Android: 
- when connecting to known device and not in range the gatt callback returns a GATT 133 error after a few seconds and we throw a connection exception.

For iOS:
- when connecting to known device and not in range, iOS will try to connect indefinitely. Use the cancelation token to cancel the request.

The feature is still not released or final so any API improvement suggestions are welcome.

As a general rule of thumb any: Async BLE plugin method calls can throw an exception in one form or another, so it is good practice to surround them in try {} catch{} blocks at some level in your app.

alberk8, 2016-08-11:
Hi,
That is exactly what I did.  I surround the ConnectToKnownDeviceAsync with a try {} catch {} block.  The system did indeed catch the exception thrown.  Immediately after the exception is caught the application will crash.

xabre, 2016-08-11:
are you awaiting the ConnectToKnownDeviceAsync call in your try catch block?

alberk8, 2016-08-11:
Yes,
```
try
{
 await _bleAdapter.ConnectToKnownDeviceAsync(Guid.Parse("00000000-0000-0000-0000-f6bfd73398f7"));
}
catch
{
  msg = "Cannot Connect to Bluetooth Device";
}
```

alberk8, 2016-08-12:
The following is my Sample Code that will always crash

```
using System;
using Android.App;
using Android.Widget;
using Android.OS;
using Plugin.BLE;
using Plugin.BLE.Abstractions.Contracts;

using Plugin.BLE.Abstractions.Exceptions;

namespace BluetoothMonitor
{
    [Activity(Label = "BleMonitor", MainLauncher = true, Icon = "@drawable/icon")]
    public class MainActivity : Activity
    {
    
        private Plugin.BLE.Abstractions.Contracts.IAdapter _bleAdapter;
        private IBluetoothLE _bluetooth;
        private Button scanButton;
        private Button stopButton;
        private EditText edText;
        private DateTime datetime;

        protected override void OnCreate(Bundle bundle)
        {
            base.OnCreate(bundle);

            _bluetooth = CrossBluetoothLE.Current;
            _bleAdapter = CrossBluetoothLE.Current.Adapter;
            _bleAdapter.ScanTimeout = 20000;
           

            // Set our view from the "main" layout resource
            SetContentView(Resource.Layout.Main);

            // Get our button from the layout resource,
            // and attach an event to it
            stopButton = FindViewById<Button>(Resource.Id.stopbutton);
           
            connButton = FindViewById<Button>(Resource.Id.connblebtn);
            edText = FindViewById<EditText>(Resource.Id.edittext);

            

            
            stopButton.Click += delegate { this.Finish(); };
          
            connButton.Click += bleconnect;

        

        }

        public void bleScan(object sender, EventArgs e)
        {
            _bleAdapter.StartScanningForDevicesAsync();
        }

       

        public async void bleconnect(object sender, EventArgs e)
        {
            string errorstr ="" ;
            TimeSpan scantime; 
            if(_bluetooth.IsAvailable)
                if(_bluetooth.IsOn)
                {
                    datetime = DateTime.Now;
                   
                    try
                    {
                        edText.Text = "Start Scanning...";
                       
                        await _bleAdapter.ConnectToKnownDeviceAsync(Guid.Parse("00000000-0000-0000-0000-ea7a9b5b504f"));
                        scantime = DateTime.Now - datetime;
                    }
                    catch(DeviceDiscoverException er )
                    {
                        errorstr = er.Message;
                    }

                   
                    scantime = DateTime.Now - datetime;
                    if (errorstr == "")
                        edText.Text = "Scan Time in ms :" + scantime.TotalMilliseconds.ToString() + "\n";
                    else
                        edText.Text = errorstr + " :" + scantime.TotalMilliseconds.ToString() + "\n";

                   
                    

                }
        }
    }
}

```

Crash Log
```
08-12 10:00:34.724 D/BluetoothManager(17049): getConnectionState()
08-12 10:00:34.724 D/BluetoothManager(17049): getConnectedDevices
08-12 10:00:34.736 D/Mono    (17049): Assembly Ref addref Plugin.BLE.Abstractions[0x558ef10990] -> System.Threading[0x558ef13b70]: 3
08-12 10:00:34.743 D/BluetoothGatt(17049): connect() - device: EA:7A:9B:5B:50:4F, auto: false
08-12 10:00:34.743 D/BluetoothGatt(17049): registerApp()
08-12 10:00:34.746 D/BluetoothGatt(17049): registerApp() - UUID=1352414f-23ea-471e-9876-6b60c8a35446
08-12 10:00:34.748 D/BluetoothGatt(17049): onClientRegistered() - status=0 clientIf=5
08-12 10:01:04.766 D/BluetoothGatt(17049): onClientConnectionState() - status=133 clientIf=5 device=EA:7A:9B:5B:50:4F
Thread started:  #2
OnConnectionStateChange: GattCallback error: 133
08-12 10:01:04.807 I/mono-stdout(17049): OnConnectionStateChange: GattCallback error: 133
Failed to connect peripheral 00000000-0000-0000-0000-ea7a9b5b504f: 
08-12 10:01:04.814 I/mono-stdout(17049): Failed to connect peripheral 00000000-0000-0000-0000-ea7a9b5b504f: 
ConnectAsync Error: 00000000-0000-0000-0000-ea7a9b5b504f 
08-12 10:01:04.821 I/mono-stdout(17049): ConnectAsync Error: 00000000-0000-0000-0000-ea7a9b5b504f 
Disconnected by user
08-12 10:01:04.862 D/Mono    (17049): DllImport searching in: '__Internal' ('(null)').
08-12 10:01:04.863 D/Mono    (17049): Searching for 'java_interop_jnienv_call_nonvirtual_boolean_method_a'.
08-12 10:01:04.863 D/Mono    (17049): Probing 'java_interop_jnienv_call_nonvirtual_boolean_method_a'.
08-12 10:01:04.863 D/Mono    (17049): Found as 'java_interop_jnienv_call_nonvirtual_boolean_method_a'.
08-12 10:01:04.865 I/mono-stdout(17049): Disconnected by user
08-12 10:01:04.886 D/BluetoothGatt(17049): close()
08-12 10:01:04.886 D/BluetoothGatt(17049): unregisterApp() - mClientIf=5
Error while connecting ''. Not raising disconnect event.
08-12 10:01:04.887 I/mono-stdout(17049): Error while connecting ''. Not raising disconnect event.
Unhandled Exception:

Plugin.BLE.Abstractions.Exceptions.DeviceConnectionException: GattCallback error: 133
```

xabre, 2016-08-12:
Try to call the method with a cancellation token, even if you set CancellationToken.None. I think there is a signature missmatch

xabre, 2016-08-12:
I'll fix it as soon as possible

alberk8, 2016-08-13:
Hi,

I have uninstall the Application, clean the solution and rebuild the app and deploy again.  Seems to work now.   I am not sure what happened.  The exception is now caught in the Application and does not crash.  Will report back if this happen again.  Thank you for helping out.

closed 2016-08-25 by xabre