dotnet-bluetooth-le/dotnet-bluetooth-le #74: StartScanningForDevicesAsync for Specific Device Not Working

alberk8, 2016-08-13:
## Steps to reproduce

1.  Scanning for Specific Device.  (no result)
    (If I replace the command with ConnectToKnownDeviceAsync - connection is ok) that means that the GUID is good.
2. 

3.

## Expected behavior
Should discover the Device with the Correct Known GUID

## Actual behavior
No Device Found

### Crashlog
NONE

## Configuration

**Version:** V1.1.0 (download from source)

**Platform:** Android 6.0.1
Visual Studio 2015 V14.0.25421.03 Update 3
Xamarin 4.1.2.18
Xamarin.Android 6.1.2.21
Android API 23

**Device:**  Xioami Redmi 3S

**Code**
```
using System;
using Android.App;
using Android.Widget;
using Android.OS;
using Plugin.BLE;
using Plugin.BLE.Abstractions.Contracts;
using Plugin.BLE.Abstractions.EventArgs;
using Plugin.BLE.Abstractions.Exceptions;
using System.Timers;

namespace BluetoothMonitor
{
    [Activity(Label = "BluetoothMonitor", MainLauncher = true, Icon = "@drawable/icon")]
    public class MainActivity : Activity
    {

        private Plugin.BLE.Abstractions.Contracts.IAdapter _bleAdapter;
        private IDevice currDevice;
        private IBluetoothLE _bluetooth;
        
        private Button connButton;
        private Button stopButton;
        private EditText edText;
        private EditText edTextRssi;
        private DateTime datetime;
     
     
        protected override void OnCreate(Bundle bundle)
        {
            base.OnCreate(bundle);

            _bluetooth = CrossBluetoothLE.Current;
            _bleAdapter = CrossBluetoothLE.Current.Adapter;
            _bleAdapter.ScanTimeout = 15000;
           

            // Set our view from the "main" layout resource
            SetContentView(Resource.Layout.Main);
            
            // Get our button from the layout resource,
            // and attach an event to it
            stopButton = FindViewById<Button>(Resource.Id.stopbutton);
            connButton = FindViewById<Button>(Resource.Id.connblebtn);
            edText = FindViewById<EditText>(Resource.Id.edittext);
            edTextRssi = FindViewById<EditText>(Resource.Id.edittextrssi);

            //button.Click += delegate { button.Text = string.Format("{0} clicks!", count++); };
            stopButton.Click += delegate { this.Finish(); };
       
            connButton.Click += bleconnect;

            _bleAdapter.DeviceConnected += bleConnected;

            _bleAdapter.DeviceDiscovered += bleDiscovered;

        }

      
     

        public void bleConnected(object sender, DeviceEventArgs e)
        {
            currDevice = e.Device;
        }

        public void bleScan(object sender, EventArgs e)
        {
            _bleAdapter.StartScanningForDevicesAsync();
        }

        public async void bleDiscovered(object sender, DeviceEventArgs e)
        {
            currDevice = e.Device;
            Toast.MakeText(this, e.Device.Id.ToString(), ToastLength.Long).Show();
            if (_bleAdapter.IsScanning)
                try
                {
                    await _bleAdapter.StopScanningForDevicesAsync();
                }
                catch
                {
                    System.Diagnostics.Debug.WriteLine("Cannot Stop Scanning");
                }
        }

        public async void bleconnect(object sender, EventArgs e)
        {
            string errorstr ="" ;
            TimeSpan scantime; 
            if(_bluetooth.IsAvailable)
                if(_bluetooth.IsOn)
                {
                    datetime = DateTime.Now;
                    try
                    {
                        edText.Text = "Start Scanning...";
                    
                        await _bleAdapter.StartScanningForDevicesAsync(new[] { Guid.Parse("00000000-0000-0000-0000-f6bfd73398f7"});
		
		       // await _bleAdapter.ConnectToKnownDeviceAsync(Guid.Parse("00000000-0000-0000-0000-f6bfd73398f7"));
                        scantime = DateTime.Now - datetime;
                    }
                    catch (DeviceConnectionException er) //catch(DeviceDiscoverException er )
                    {
                        Toast.MakeText(this, er.Message, ToastLength.Short).Show();
                    }
                    catch(Exception er)
                    {
                        //errorstr = er.Message;
                        Toast.MakeText(this, er.Message, ToastLength.Short).Show();
                        errorstr = "Error Occured";
                    }

                    

                    scantime = DateTime.Now - datetime;
                    if (errorstr == "")
                        edText.Text = "Scan Time in ms :" + scantime.TotalMilliseconds.ToString() + "\n";
                    else
                        edText.Text = errorstr + " :" + scantime.TotalMilliseconds.ToString() + "\n";

                   

                    
                }
        }
    }
}

```

smstuebe, 2016-08-13:
Please read carefully what this function is doing:

https://github.com/xabre/xamarin-bluetooth-le/blob/master/Source/Plugin.BLE.Abstractions/Extensions/AdapterExtenstion.cs#L26

alberk8, 2016-08-15:
Hi,
Thanks for highlighting my. mistake.

closed 2016-08-15 by alberk8