dotnet-bluetooth-le/dotnet-bluetooth-le #76: CharacteristicBase class should override Equals and Hashcode

sumitgouthaman, 2016-08-18:
The CharacteristicBase class does not override the Equals and Hashcode method. Realized that when I was using a HashSet to keep track of Characteristics.

I guess using the internal Guid as the basis for the Equals and Hashcode implementation should work.

smstuebe, 2016-08-18:
Ok thanks, but we should do it for all objects :)

sumitgouthaman, 2016-08-18:
I can take a crack at doing this

adolgov, 2016-08-18:
Wouldn't using the Guid make all characteristics from similar devices be equal? i.e. if I have multiple sport trackers that have the same service and characteristic profile, each of the "write" characteristics would be seen as equal, even though their behavior would be very different (data would be written to different physical devices).

I think that what would uniquely identify a characteristic (for equality purposes) would be the set of:

- Guid of Characteristic
- Guid of its Service (this one is least needed, actually. See below)
- Guid of Device

Usual convention could probably get rid of the Service, since most services have characteristics with unique IDs in them. When using a short characteristic ID, the ID gets plugged into the long service ID, making it unique. However, as far as I know, it's not _illegal_ to use a custom long characteristic ID, and to use that same custom long characteristic ID in different services, though I have not seen this ever happen.

I think that the breakdown of IEquatable is similar to:

- Device -> Guid of device (I remember writing this one up either late last year or early this year, so it might already exist in the code.
- Service -> Guid of device + Guid of Service
- Characteristic -> as paragraph above

sumitgouthaman, 2016-08-18:
@adolgov : You are right. The hierarchy needs to be taken into account.