dotnet-bluetooth-le/dotnet-bluetooth-le #77: Enhancement: IsEnabled for Service and Characteristic

alberk8, 2016-08-18:
I would be good to have an additional `IsEnabled` property for `GetServiceAsync` and `GetCharacteristicAsync`.

Use Case:
I have a `System.Timers.Timer` loop in the background to query for data from BLE devices (more than one device type).  The issue is that not very BLE device are make equal,  Sometimes, when the `GetServiceAsync` and `GetCharacterisAsync` are called and returns without any exceptions but it is instance the service and characteristic is not wired up.  So if I call the `WriteAsync() `,  it will throw an exception.  I would be good if I can test the condition before calling.    

```
try{
_Service = await _Device.GetServiceAsync(Guid.Parse(ServiceGuid));
_Characteristic = await _Service.GetCharacteristicAsync(Guid.Parse(RWGuid));
_NotifyCharacteristic = await _Service.GetCharacteristicAsync(Guid.Parse(NotifyGuid));
await Task.Delay(150);
}
catch(Exception ex)
{
  Log.Info(TAG,"Error "+ex.Message);
}
//
try{
await _Characteristic.WriteAsync(cmd);
}
catch(Exception ex)
{
  Log.Info(TAG,"Error Writing"+ex.Message);
}

```

smstuebe, 2016-08-18:
What should `IsEnabled` do?

alberk8, 2016-08-18:
`IsEnabled ` should be a bool to indicate if the underlying service/characteristic is wired up and active.

smstuebe, 2016-08-18:
What does this mean? Which properties/states indicate the states `wired up` and `active`?

xabre, 2016-10-20:
I'm guessing this issue is kind of stale ...

closed 2016-10-20 by xabre