dotnet-bluetooth-le/dotnet-bluetooth-le #79: BLE Bonding and SetPin

alberk8, 2016-08-23:
Hi,
I understand that bonding and setpin is not yet implemented .  May I know how soon is this going to be implemented and How do i go about getting this to work in the meantime (quick and dirty way :) ).   Thank you.

smstuebe, 2016-08-23:
Not planned yet and we have no time plan, because bug fixing and enhancing/stabilizing existing features is the current focus. You could use the source of the plugin and implement it on your own.
see #32

alberk8, 2016-08-23:
Hi,
Thanks for the reply.   With regards to the #32 , Is in the v1.1 or v1.0 code base?  Thanks you.

smstuebe, 2016-08-23:
1.1 is current development branch.
1.0 is latest stable

alberk8, 2016-08-24:
Hi,
I have checked to source code  in V1.0 and 1.1 and could not find the any implementation of the BLE bonding for Android as per #32.  Any help would be much appreciated.  Thank you.

smstuebe, 2016-08-24:
#32 is a task with the label "investigation needed". This means: That we have to clarify how this feature has to be implemented on each platform, first. Sorry if the text of the issue is misleading. There was a implementation but it was out of scope.

alberk8, 2016-08-24:
Yes, the text was quite misleading and thanks for the clarification.  I have a few devices, Glucometer and Blood Pressure Monitor that I can test for bonding and/or MITM.   I would gladly test the plugin even in alpha form.  Hope that would come soon.  Thank you.

closed 2016-08-24 by smstuebe