dotnet-bluetooth-le/dotnet-bluetooth-le #80: Question. Can I use other ble library for some reason to update firmware with using IDevice class?

allbdeveloper, 2016-08-27:
Hi, I'm making app with using Xamarin.forms PCL. 
My pcb is powered by nordic's nRF51822 MCU. 
They provided great example source code for DFU OTA(update its firmware on to air) but it's only for iOS and Android native. 

https://devzone.nordicsemi.com/question/89049/dfu-in-ios-xamarin/?comment=92368#comment-92368
This guy binded its DFU library to c# but they made it as depended library to their BLE library which called Arendi BLE library. 

It's not free license but they gave me source code for DFU. It's free. 
But problem is to call DFU update method, I have to put First parameter as it's IPeripheral class. 
What the heck? 

What I'm curious is that is there any way to use xabre's IDevice class to call that method?
Basicly some part might be same on each native.

xabre, 2016-08-29:
I am not familiar with their library but if the concept of IPeripherial is similar to our IDevice abstraction you might not be able to get an instance of both, in our library to get a IDevice instance after scanning or connecting. In version 1.1 you are able to get a connected IDevice by calling ConnectKnwonDeviceAsync (GUID). 

My thoughts on this are the following: 
- try to get the IPeripherial using their library (connect I assume)
- try to get an IDevice by calling ConnectKnownDeviceAsync (v.1.1) with the GUID of the IPeripherial
- Build you own custom Device class which has a reference to IDevice and IPeripherial...
- Figure out how to handle disconnection (which library you use to call it)...

allbdeveloper, 2016-08-29:
Thanks for your answer.

closed 2016-09-08 by smstuebe