dotnet-bluetooth-le/dotnet-bluetooth-le #81: iOS app crash suddenly with "Failed to marshal the Objective-C"

allbdeveloper, 2016-09-06:
## Steps to reproduce

1. Suddenly...

## Actual behavior
App Crash Suddenly

### Crashlog

Failed to marshal the Objective-C object 0x14f0b340 (type: CoreBluetooth_CBPeripheral__CBPeripheralDelegate). Could not find an existing managed instance for this object, nor was it possible to create a new managed instance (because the type 'CoreBluetooth.CBPeripheral+_CBPeripheralDelegate' does not have a constructor that takes one IntPtr argument).

## Configuration

**Version:** 1.0
**Platform:** iOS
**Device:** iPhone 5

smstuebe, 2016-09-07:
which app? iOS Version?

allbdeveloper, 2016-09-07:
Hi, 
My iPhone 5's iOS version is 9.3.2(13F69) 
I will check it together. I have been using it from nuget so far, I'm going to get plugin from github.

smstuebe, 2016-09-07:
And which App? Your app? The Sample App?

allbdeveloper, 2016-09-07:
Oh, It's my app.

jasonchester, 2016-09-09:
I got this error as well when I was in the office with a lot of Peripherals. I assume it might be a specific one is causing issue or somehow a device may be getting turned off at some time before the app could process the event.

commenting out the smelly code made this go away.

_Plugin.Ble.Ios_@Line 24
        public Device(Adapter adapter, CBPeripheral nativeDevice, string name, int rssi, List<AdvertisementRecord> advertisementRecords) : base(adapter)
        {
            _nativeDevice = nativeDevice;
            Id = Guid.ParseExact(_nativeDevice.Identifier.AsString(), "d");
            Name = name;

            Rssi = rssi;
            AdvertisementRecords = advertisementRecords;

            // TODO: smell!!!! WeakSubscribe?!
            //_nativeDevice.UpdatedName += (sender, e) =>
            //{
            //    Name = ((CBPeripheral)sender).Name;
            //    Trace.Message("Device changed name: {0}", Name);
            //};
        }

allbdeveloper, 2016-09-10:
Thanks Jason and smstuebe. 
Can you provide nuspec file? I really have to try with fixed 1.1.0. 
But some reason if I use this as library, Realm cause problem. Realm team is looking the problem. (I think it's realm's problem)

But when I use both with nuGet package, there was no problem. So I think I should create nuget package and use it locally. 

So if I can download nuspec file, it's really helpful. 
(making nuspec file is easy? I didn try before)

Thanks.

xabre, 2016-09-22:
Removed code smell in 1.1.0-beta2

closed 2016-09-22 by xabre