dotnet-bluetooth-le/dotnet-bluetooth-le #83: GattCallback error : 133 on Android.

allbdeveloper, 2016-09-10:
## Steps to reproduce

App crashes suddenly. 

> "GattCallback error : 133 on Android."

(I guess it happens when it's getting low signal because it's little bit far from peripheral)

### Crashlog

## Configuration

**Version:** 1.0

**Platform:** Android

**Device:** Samsung Galaxy S4 (Lollipop)
[image]

allbdeveloper, 2016-09-10:
I'm having this issue many times.

allbdeveloper, 2016-09-11:
I put 
gatt.close(); after disconnect(); 

problem seems gone so far.

smstuebe, 2016-10-21:
We will look at  https://github.com/myallb/xamarin-bluetooth-le ^^

AnoopChatra, 2017-01-06:
I am still facing this problem many times.
I am able to bond/pair with the BLE device but when I try to connect using 
"Cross adapter.ConnectToDeviceAsync(BLEdevice)"  I am getting the following exception
"Plugin.BLE.Abstractions.Exceptions.DeviceConnectionException: GattCallback error: 133"

Device used : Samsung S7 android 6.0(marshmallow)
plugin version : Plugin.BLE.1.2.0-beta3

AnoopChatra, 2017-01-11:
Does anyone has solution to this problem?

AnoopChatra, 2017-01-11:
@allbdeveloper  , How to access "gatt" object ?? It is been abstracted by plugin .

allbdeveloper, 2017-01-11:
You can't access it unless clone git project and build your own.

By the way, I recommend you to use try-catch in the place where you have a problem. 
It worked for me and problem is gone.

oleg-123, 2017-04-21:
Periodically there is such problem on a meizu pro6. I use a package 1.1.0

chapeti, 2017-04-21:
Same error here

HenrySipp, 2017-05-12:
Same error here

erdmenchen, 2017-06-22:
I get to see this error every now and then, too. 
Is the missing gatt.close(); after the disconnect(); a problem for future connecting?

kuropone, 2018-01-18:
i still got this error on my device.
always error when try to connect device.

kobin97, 2021-02-03:
var parameters = new ConnectParameters(true, true);
 await adapter.ConnectToDeviceAsync(currentDevice, parameters, new CancellationTokenSource().Token);

kobin97, 2021-02-03:
var parameters = new ConnectParameters(true, true);
  await adapter.ConnectToDeviceAsync(yourDevice, parameters, new CancellationTokenSource().Token);

rajnikantpatel23, 2021-04-29:
I have resolved my Android issue using below code when connecting any device.

var parameters = new ConnectParameters(forceBleTransport: true);
await _adapter.ConnectToDeviceAsync(deviceToConnect, parameters, cancellationToken.Token);

Let me know if it works :)