dotnet-bluetooth-le/dotnet-bluetooth-le #84: Backgroundtasks

DumDumin, 2016-09-12:
Hey guys,
Is your library able to run and receive ble notifications while the app is in the background?
If not are you planning in implementing that feature?

Kind Regards

allbdeveloper, 2016-09-12:
Hm strange. 
I wanted to know why 'xabre' added "needs investigation" label. 

As I know, this library is working in the background. 
To be clear, it's not the library but native library is created to this way.

smstuebe, 2016-09-12:
This needs investigation, because we haven't tested it and we have no implementation in the sample app that proves it ^^ And we haven't read a lot about the restrictions and requirements on each platform regarding the background ble.

allbdeveloper, 2016-09-12:
OK. I see. Thanks for reply!

DumDumin, 2016-09-13:
Well yeah, every native platform has its own background implementation. Windows has the background tasks with a GattCharacteristicNotificationTrigger and Android has its Services which can run in the background, dunno what iOS can do ;) what I know so far is, that apps wont run in the background for long, if they dont have any permission to do so. This permission can be achieved by register "backgroundtasks" for bluetooth or other external sources. Thats what i mean, register backgroundtasks. Please correct me if i got something wrong, but i dont understand why this library should run in the background without any registration for it.

xabre, 2016-09-14:
@DumDumin @allbdeveloper This is exactly why I flagged it as 'needs investigation'.

smstuebe, 2016-10-21:
@xabre -> add to docs