dotnet-bluetooth-le/dotnet-bluetooth-le #85: Bonding State of a BLE device

leelatoo, 2016-09-15:
Hi,

1. How to know the bonding state of a discovered BLE device?
In the plugin, i came across the API "DeviceBondState". What is the use of it and how to use it?
2. What is the difference between connection state and bonding state?

xabre, 2016-09-15:
Bond state comes from the Android API, a long time ago this was a feature for the android part of the library. Since the refactoring and stabilisation of the library this has not been a priority yet. We will implement this as a feature as soon as we find the corresponding feature on the iOS side and find the time to actually do it.

For more information on bonding/pairing in Android check this link:
[https://developer.android.com/reference/android/bluetooth/BluetoothDevice.html#createBond()](https://developer.android.com/reference/android/bluetooth/BluetoothDevice.html#createBond())

leelatoo, 2016-09-15:
Ok. Thanks for your answers

leelatoo, 2016-09-15:
Is there a way to know the whether Bluetooth LE device is paired with the smartphone?

xabre, 2016-09-19:
For android check: [https://developer.android.com/reference/android/bluetooth/BluetoothDevice.html#getBondState()](https://developer.android.com/reference/android/bluetooth/BluetoothDevice.html#getBondState())

As I said, this is not yet part of our cross platform library.

allbdeveloper, 2016-10-11:
I also wanna say that it should be optional I think. 
Some people wants to make bond state to disable. (I'm too)

And question, Sometime on iOS, my peripheral went into bond state (it's connected on iOS bluetooth setting). How could I disable it? 

Once it's connected to specific iPhone, It means it can not be connected with other phone.

Thanks.

simson51, 2017-02-22:
I guess we have to wait for secure communication with BLE devices until both, Mr. Android and Mrs. Apple decide to enable AppDevs to do so.

benevbright, 2017-02-27:
My question was wrong and it was my mistake. 

Thanks.

asierdanjou, 2018-11-30:
Hi,
What about bond state querying in XF ? any advance on it ?
If there is no advances, how can I implement it in android project ? 
Thanks and best regards.

mallikarjunpatala, 2019-07-09:
Hi,
How to clear bonding information in BLE after paring.