dotnet-bluetooth-le/dotnet-bluetooth-le #87: Cast CBPeripheral to IDevice

leelatoo, 2016-09-16:
Hi,

In Xamarin.iOS project, I get connected BLE devices as CBPeripheral objects. I want to use these objects with Plugin.BLE APIs. Since Plugin.BLE accepts the BLE devices as IDevice, casting CBPeripheral to IDevice is required.

Is it possible to cast CBPeripheral to IDevice?

Similarly, for Android also, is it possible to cast native android bluetooth LE device to IDevice?

Thanks

xabre, 2016-09-16:
The whole point of the plugin is to to this for you. The plugin does this internally and provides an abstraction (IDevice). 'Casting' CBperipherial and BluetoothDevice to IDevice implies actually doing the work of the plugin by yourself. 
Does the plugin lack some of the native functionality you require? Can you explain your use-case more in detail?

leelatoo, 2016-09-16:
Yes, my requirement is to get the BLE devices which are already connected to android/iOS system.
So, i use retrieveConnectedPeripherals in iOS and GetConnectedDevices in Android.

After retrieving the connected devices using platform specific code, i want to use these CBPeripheral/BluetoothDevice objects with Plugin.BLE. This is the scenario i am dealing with.

Is there any API in Plugin.BLE which gives connected BLE devices?

xabre, 2016-09-16:
I have been thinking about this functionality for a while now. I will mark this as an enhancement.

xabre, 2016-09-21:
Check #86. Marked as duplicate.

closed 2016-09-21 by xabre