dotnet-bluetooth-le/dotnet-bluetooth-le #88: How to use the Nordic UART Service

machohv, 2016-09-19:
I use this Nugget to connect my Crossplatform App with a Bluefruit Feather. I programmed the Bluefruit device to receive and send a message. With the "UUID 6E400001-B5A3-F393-E0A9-E50E24DCCA9E" I get the Nordic UART Service and the TXCharacteristic with "6E400002-B5A3-F393-E0A9-E50E24DCCA9E" and the RXCharacteristic with "6E400003-B5A3-F393-E0A9-E50E24DCCA9E".
The TXCharacteristic has CanWrite = true, CanRead = false, CanUpdate = false
The RXCharacteristic has CanWrite = false, CanRead = false, CanUpdate = true
I thought that by writing data in the TXCharacteristic I would read the answer in the RXCharacterisitic but I don't know how to make it work

xabre, 2016-09-19:
Im assuming that you have to subscribe to notifications on the RXCharacteristic and start notifcations for it. Once you write something on TX a notification event will return on RX with the result.

machohv, 2016-09-19:
How do I subscribe for notifications on a characterisitic? 
All I did with the RXCharacterisitic was to copy the code of the notifications and the use the WriteAsync in the TXCharacteristic 

characteristic.ValueUpdated += (o, args) =>
{
    var bytes = args.Characteristic.Value;
};

characteristic.StartUpdates();

xabre, 2016-09-19:
So this does not work? What does your Output say (most send/receive events are traced to the console by the library)?

machohv, 2016-09-19:
I successfully write in the TXCharacteristic but the value of the RXCharacteristic never changes, this is the part of my code when I try to use the service

[image]

xabre, 2016-09-21:
Any news? Did you manage to get it working? What does your output log say?

DumDumin, 2016-10-08:
What are you trying to achieve here? Where is your code of your microcontroller with the bluefruit? maybe there is something going on.

DumDumin, 2016-10-08:
BTW, on Android I got the issue that the characteristic.StartUpdates() only starts updating in about 50% of the cases. I have no clue why and couldnt get any idea from your characteristic implementation.

What I did to solve this is to use await Task.Delay(2000); and call StartUpdates() again.

xabre, 2016-10-08:
@DumDumin I've also noticed this start notifications issue. The cause seemed to be not waiting for the descriptor write gatt callback and the immediatly executing a new ble command. It's been fixed now with the introduction of Descriptor read.write async operations.

smstuebe, 2016-10-21:
@DumDumin Can you please try it with the latest release beta5 or 1.1.0.

    await characteristic.StartUpdatesAysnc();

otherwise we will close this issue after 8 days without response

CarlosRodrigoCode, 2017-01-06:
@machohv Hi! Hola macho. Did you fix this issue with the BLE plugin in xamarin forms? I have the same BLE device and i am having trouble reading the correct RX value that is being sent from the BLE device. If you solved it i would be very happy to see how you did it ! :)

closed 2017-01-08 by smstuebe

Duality4Y, 2017-06-07:
@machohv I am verry interested in your code, because I too am working on bluetooth communication with a nordic device.

machohv, 2017-06-07:
At the end my code worked like this
[image]

Duality4Y, 2017-06-07:
I was hoping for a bit more code, like how you setup the bluetooth and everything :)

Duality4Y, 2017-06-08:
What exactly does this code do? does it send back what it receives ?