dotnet-bluetooth-le/dotnet-bluetooth-le #92: bluetoothManager.GetConnectedDevices(ProfileType p) always returns an empty list

AnoopChatra, 2016-09-27:
Hi,

          I am trying to use bluetoothManager.GetConnectedDevices(ProfileType p) method to retrieve all the connected BLE devices. I have tried passing "GATT" and "GATT_SERVER" as profile types. 
     I have already paired with a BLE device in default android settings page.Now i am expecting this method to return a list of BLE devices which are already paired but it is always returning an empty list.  

## Expected behavior
The method should return a list of Paired/Connected BLE device list when "GATT" or "GATT_SERVER" passed as profile types

## Actual behavior
Returning an empty list

##Reference
#86 says list of connected BLE devices can be retrieved using GetConnectedDevices() native android API

## Configuration

**Version:** android 6.0

**Platform:** Android/ ...

**Device:** nexus 5

xabre, 2016-09-28:
@AnoopChatra  Well this is actually expected on android. A paired/bonded device is not necessary connected.  For example I paired my fitness tracker and left it home. At work I can still see it in my paired device list in android but it's surely not connected. 

On android, GetConnectedDevices will only return the physically connected devices, i.e. devices that are connected by other apps.

Android offers a different API call for getting the paired/bonded device. This call is not yet included in our plugin. 
https://developer.android.com/reference/android/bluetooth/BluetoothAdapter.html#getBondedDevices()

xabre, 2016-10-03:
#98

xabre, 2016-10-14:
Try it with 1.1.0-beta5

closed 2016-10-14 by xabre