dotnet-bluetooth-le/dotnet-bluetooth-le #96: Android 4.4 StartScanningForDevicesAsync() no results

andrew-nemtsev, 2016-09-29:
The following issue is observed for Vanilla Plugin.BLE:
DeviceDiscovered event is triggered for Android 5.x and 6.x, but not for Android 4.4

Example project
https://github.com/andrew-nemtsev/plugin_ble_android4_issue

## Steps to reproduce

1. Start app @ Android 4.4 device in range of any BLE-advertising device

2. Press the button

## Expected behavior
"DISCOVERED" toast is appeared

## Actual behavior
"DISCOVERED" toast is NOT appeared

## Configuration

**Version:** 1.0.0 and 1.1.0 beta-2

**Platform:** Android 4.4.2

**Device:** Fly Evo Tech 3 IQ4414 Quad

xabre, 2016-10-03:
What doe LogCat say? Are there any errors/warnings in the output?
I have tested the lib with a Samsung Galaxy S3 Neo (4.4) and it works fine.

smstuebe, 2016-10-21:
No response within 3 weeks -> closed :facepunch:

closed 2016-10-21 by smstuebe