dotnet-bluetooth-le/dotnet-bluetooth-le #97: Create IDevice object from a CBPeripheral/Guid

leelatoo, 2016-09-30:
Hi,

We are working on an app which connects to BLE hearing aids. Requirement is to connect the hearing aids to iOS device by nagivating to the Setings->General->Accessibility->HearingAids. .Once hearing aid is connected to iOS device, hearing aid stops advertising.

The app needs the connected hearing aid BLE device objects. So, we use mono touch CorBluetooth API retrieveConnectedPeripherals(), which gives the connected hearing aids as CBPeripheral objects. We want the app to connect to the CBPeripheral objects using Plugin.BLE.
But Plugin.BLE requires BLE devices as IDevice objects. So, the CBperipheral objects has to be converted to IDevice objects. Currently Plugin.BLE does not provide any API for CBPeripheral to IDevice conversion.

We are in need of the conversion facility by Plugin.BLE. The need of this conversion functionality is already mentioned in https://github.com/xabre/xamarin-bluetooth-le/issues/87. It is added to Plugin.BLE 1.10 milestone.

When can we expect the release of Plugin.BLE version 1.1.0?

Or is there any alternative solution to get an IDevice object out of CBPeripheral object? How can we create an Device object from a CBPeripheral or a GUID?

xabre, 2016-09-30:
A few questions:
Did you update to 1.1.0-beta3?
Did you try calling:
- **GetSystemConnectedDevices** and then **ConnectAsync**
OR
- **ConnectToKnownDeviceAsync**(guid)

leelatoo, 2016-10-03:
No, we are using 1.0.0 beta 3.

we updated the nuget package to 1.1.0 beta4, but it does not get compiled. we are using Xamarin version 5.10.1(build 6), Xamarin.iOS 9.4.1.25, Mono 4.4.0

**Compile time error:** 
_MTOUCH: Error MT2002: Failed to resolve "!!0[] System.Array::Empty()" reference from "mscorlib, Version=2.0.5.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e" (MT2002)_

To work with nuget package to 1.1.0 beta3, latest Xamarin version is required?

xabre, 2016-10-03:
All my dev machines are up to date so it's difficult to say what the dependencies may be and what the cause of your build error is. We always use the clould CI service from bitrise to confirm that the build is not just 'working on my machine'. 
Is there anything blocking you from updating to the latest xamarin version? 
@smstuebe any ideas?

Alternatively you can check out the library sources from the 1.1.0 branch build it yourself and reference the dll's manually in your project.

leelatoo, 2016-10-04:
Hi,

We have tried this option
_Alternatively you can check out the library sources from the 1.1.0 branch build it yourself and reference the dll's manually in your project._

We get the same compile time error :(

leelatoo, 2016-10-04:
Additionally, we checked upgrading to Xamarin studio 6.1.1, Xamarin.iOS, 10.0.1.8, Xcode 8.0 and updated the nuget package. 

**GetSystemConnectedDevices** call always return empty list (though BLE devices are connected to the iOS system).

xabre, 2016-10-04:
@leelatoo I will need your help on this one. 
Currently I don't have access to a ble device with this capability/use case so it's difficult to test. I will need you to show me a code sample of how you call the iOS method (_centralManager.RetrieveConnectedPeripherals)? 

Currently the iOS implementation in the library looks like this, maybe there is some problem with it:

````
  public override List<IDevice> GetSystemConnectedDevices(Guid[] services = null)
        {
            var serviceUuids = new CBUUID[0];
            if (services != null)
            {
                serviceUuids = services.Select(s => CBUUID.FromString(services.ToString())).ToArray();
            }

            var nativeDevices = _centralManager.RetrieveConnectedPeripherals(serviceUuids);

            return nativeDevices.Select(d => new Device(this, d)).Cast<IDevice>().ToList();
        }
```

leelatoo, 2016-10-05:
Sure.

This is the piece of code which we are using.

private static IBluetoothLE ble = CrossBluetoothLE.Current;
private static IAdapter adapter = CrossBluetoothLE.Current.Adapter;

public override void ViewDidLoad ()
{
	base.ViewDidLoad ();
	string guidStr = "C00B182A-D0R0-1000-82G3-00805B8F65FB";
	Guid[] serviceGuids = new Guid[1] {Guid.Parse (guidStr)};
	IList<IDevice> connectedDevices = adapter.GetSystemConnectedDevices (serviceGuids);

	Console.WriteLine ("number of Connected devices: "+connectedDevices.Count);
	for (int i = 0; i < connectedDevices.Count; i++) 
	{
	        Console.WriteLine (i+" : "+connectedDevices[i].Name+" - "+connectedDevices[i].Id);
	}
}

Number of connected devices is always 0 (though BLE devices are connected to the iOS system).

xabre, 2016-10-05:
Do you get any results when you aren't passing the GUIDs?

Just by calling:

```List<IDevice> connectedDevices = adapter.GetSystemConnectedDevices ();```

leelatoo, 2016-10-06:
Yes, We tried **adapter.GetSystemConnectedDevices ()**, the count is still 0 :(

```
IList<IDevice> connectedDevices = adapter.GetSystemConnectedDevices();

Console.WriteLine("number of Connected devices: " + connectedDevices.Count);
for (int i = 0; i < connectedDevices.Count; i++)
{
	Console.WriteLine(i + " : " + connectedDevices[i].Name + " - " + connectedDevices[i].Id);
}
```

xabre, 2016-10-06:
> So, we use mono touch CorBluetooth API retrieveConnectedPeripherals(), which gives the connected hearing aids as CBPeripheral objects.

@leelatoo Can you show me your CoreBluetooth code for doing this. As I mentioned before, the library calls this exact function in the background so I want to see the differences.

leelatoo, 2016-10-06:
Sure, this is the CoreBluetooth API we are using.

```
CBCentralManager _central = new CBCentralManager (DispatchQueue.CurrentQueue);
CBUUID serviceUuid = CBUUID.FromString ("C00B182A-D0R0-1000-82G3-00805B8F65FB");
CBUUID[] identifiers = new CBUUID[]{serviceUuid};
CBPeripheral[] peripherals = _central.RetrieveConnectedPeripherals(identifiers);
Console.WriteLine ("Connected peripherals : "+peripherals.Length);

for (int i = 0; i < peripherals.Length; i++) 
{
	Console.WriteLine (i+" : "+peripherals[i].Name);
}
```

this piece of code returns the connected peripherals as 1 and prints device name.

xabre, 2016-10-07:
@leelatoo I found the bug and fixed it. 
Please try the implementation from 1.1.0-beta5 and let me know if ti works. The method is now called GetSystemConnectedOrPairedDevices(guid[]). I also found out that it's mandatory to specify a service guid, without that the method will always return an empty list.

leelatoo, 2016-10-12:
@xabre We updated the nuget package to 1.1.0-beta5 and tried the new API GetSystemConnectedOrPairedDevices(guid[]). It works:) returns the connected peripherals.

Thanks a lot :)

commit c095d1e9 (2016-10-07): Merged #98 & #86 into one abstraction. As much as possible.
Fixed #97. Cause was wrong paramter fo service CBUUID in ios implementation.
closed 2016-10-12 by xabre