dotnet-bluetooth-le/dotnet-bluetooth-le #101: Manufacturer Specific Data iOS

havand12, 2016-10-12:
Hi,

I've understood that iOS hides some of the advertised data from iBeacons. 

On Android the plugin gives me access to all the information I want. UUID, Major, Minor, TxPower, Rssi etc, but on iOS there is no ManufacturerData from AdvertisementRecords. 

How can I use this plugin to get information about the iBeacon on iOS? 

- Thanks :) 

## Configuration

**Version:** 1.0.0

**Platform:** iOS

**Device:** iPhone 5s

xabre, 2016-10-17:
@havand12 apparently iOS uses CoreLocation instead of CoreBluetooth to work with iBeacons. Currently we don't have anything in our iOS implementation for that. Would be very helpful if you could do some research on the matter and post your findings here. We can than decide how to proceed.

rogihee, 2017-02-10:
Upon connection there is an AdvertisementData property on the CBDiscoveredPeripheralEventArgs

This is an NSDictionary. The Manufacturer specific data is in the key
kCBAdvDataManufacturerData

`var manData = e.AdvertisementData.ObjectForKey(CBAdvertisement.DataManufacturerDataKey);`

xabre, 2017-04-25:
` if (key == CBAdvertisement.DataManufacturerDataKey)
                {
                    var arr = ((NSData)advertisementData.ObjectForKey(key)).ToArray();
                    records.Add(new AdvertisementRecord(AdvertisementRecordType.ManufacturerSpecificData, arr));
                }`

Is this not working?

flyingxu, 2017-09-01:
As indicated by your first sentence,  _iOS hides some of the advertised data from iBeacons_. To be more accurate, iOS hides **all** raw information from iBeacons. You will not see the raw bytes of the manufacturer data but CoreLocation's callback.
For iBeacon data (company as 0x004c, type as 0x02, 0x15), the code above will not get the desired manufacturer data on iOS by CoreBluetooth.  
You will not get the manufacturer data by CoreLocation either because CoreLocation does not provide API to get the raw data buy only "ranging" events.

If you also develop the Bluetooth device, make it a **general beacon** device but not an iBeacon device, then the code above will work as expected.

thenewguy133, 2020-10-10:
Any update on this? I need to get the major/minor on iOS but there is no manufacturing advertisement in the list of advertisements.