dotnet-bluetooth-le/dotnet-bluetooth-le #103: Multiple notifications don't work

gaminn, 2016-10-17:
Device LG G3, Android API 23.

```
characteristics = new List<Plugin.BLE.Abstractions.Contracts.ICharacteristic>();
characteristics.Add(...);
characteristics.Add(...);
characteristics.Add(...);
characteristics.Add(...);
characteristics.Add(...);
characteristics.Add(...);

foreach (var characteristic in characteristics)
{
    characteristic.StartUpdates();
    characteristic.ValueUpdated += Characteristic_ValueUpdated;
    await Task.Delay(150);
}
```

First call of StartUpdates returns:

`Characteristic.StartUpdates, successful: True`

The rest of the calls return:

`Characteristic.StartUpdates, successful: False`

I found out that adding delay in the code:

`foreach (var characteristic in characteristics)
{
    characteristic.StartUpdates();
    characteristic.ValueUpdated += Characteristic_ValueUpdated;
    await Task.Delay(150);
}`

"fixes" the issue. The delay must be at least 150 ms.

smstuebe, 2016-10-17:
Can you use the latest beta version and use `StartUpdatesAsync()`??

`await` characteristic.StartUpdatesAsync();`

xabre, 2016-10-17:
@gaminn This should be fixed in the latest beta. We were not waiting for the descriptor write callback. This added a breaking change and ```StartUpdates()``` was replaced by ```StartUpdatesAsync()```

gaminn, 2016-10-17:
Thanks guys! Beta fixes this.

closed 2016-10-17 by gaminn