dotnet-bluetooth-le/dotnet-bluetooth-le #104: No Devices found on Android 6 or higher

DumDumin, 2016-10-17:
On Android 6 and 7 there is no device found. Android 4 and 5 are working fine.
Does anyone else has this problem?

xabre, 2016-10-17:
@DumDumin 

From our readme : )
Add these permissions to AndroidManifest.xml. For Marshmallow, please follow [Requesting Runtime Permissions in Android Marshmallow](https://blog.xamarin.com/requesting-runtime-permissions-in-android-marshmallow/) 

```xml
<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
<uses-permission android:name="android.permission.BLUETOOTH" />
<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
```

Android >=6 requires location access and needs the user to grant this permission. Follow the link to see how this is done. In the future this might become part of the plugin, but for now this is up to the developer.

AmperayaniMadan, 2016-10-20:
I too am facing a similar issue. Does not discover the BLE device in Android 6.0. I have not tested in earlier versions.

I also tried the above mentioned permissions, does not work even then.

One thing i noticed is, outside of my application, when I turn on bluetooth I am able to see my BLE device in the result list. But why does it not work in my app?

DumDumin, 2016-10-20:
For me its only working if GPS is turned on, is that right?

AmperayaniMadan, 2016-10-20:
Glad it is working for you after turning on the GPS.

Its not working for me, even when the GPS is turned on nothing happens, the OnDeviceDiscovered() is not being hit. I am searching for a specific device using it UUID.

Im running on Samsung Galaxy Tab S2 with OS 6.0.1

xabre, 2016-10-20:
@AmperayaniMadan If you don't want to implement the permission requests like described in [Requesting Runtime Permissions in Android Marshmallow](https://blog.xamarin.com/requesting-runtime-permissions-in-android-marshmallow/),  you have to manually turn it on for your app form Settings>Apps>YourApp>Permissions. I've attached a screenshot from our ble sample app:

[image]

@DumDumin That is correct, Android >=6 need location to be enabled for ble scanning. This is because of beacons. :P You can read more about this on the android official doc.

AmperayaniMadan, 2016-10-20:
@xabre 
I did check the location and BT permissions in the App Manager for my App, they are enabled.

But what is curious is, turning on the BT manually, outside of my App, i am able to see my BLE device in the list, at this instant the GPS is not turned on. I tripple checked this.

:| :|

xabre, 2016-10-20:
@AmperayaniMadan are you sure it's a bluetooth low energy device (GATT) device? Maybe you can install the 'BLE Scanner' app from the play store and see if it can detect your device?

AmperayaniMadan, 2016-10-20:
@xabre
Yes it is a GATT device, I am able to see it in the BLE Scanner app and also connect and see its characteristics.

My code works fine for iOS, but when it comes to Android it flops. May be I am missing something apart from the permissions. I will share my code, if anyone is kind enough to point out what I am missing that will be very helpful.

But, can you share any sample code which works for Android 6.0.1. I can quickly try that and rectify.

Thanks

xabre, 2016-10-20:
Well the sample app works fine on my Android 6 phone together with the screenshot above :).
Just post your code snippet here :) and I'll try to help.

AmperayaniMadan, 2016-10-21:
Attaching a snippet of the method which I use to connect to and get data from my BLE device.

Will provide the output log in sometime.

[image]

xabre, 2016-10-23:
Hmm... you are calling StartScan with a guid as a parameter. You have to make sure that the peripheral includes the service guid in it's advertisement package. 
As a test, try removing the GUID parameter and let the scan run without any parameters.

AmperayaniMadan, 2016-10-24:
:), I tried that too earlier, didn't work. But my colleagues have a different model of the same Samsung Galaxy Tab S2, and they are able to connect to the BLE device. What I got to know is in their model they do not have a provision to turn off the GPS (weird !!, no privacy eh?).

But that should not be any different than manually turning on the GPS !!

We haven't implemented the runtime permissions in our app (but it should not be of any effect until you revoke the permissions right ?).

xabre, 2016-10-26:
- What does you application output window say? If the permission is not ok it will be reflected there.
- Did you try the sample app? (you also have to grant it location permissions)

PawelGasiorowski, 2017-09-05:
I have the same problem. In my case I can connect the device on both Android and IOS systems, hovewer Android does not contain the device name which is null. Galaxy S7.

AbanoubSedrakDev, 2018-02-06:
Device Name Null while address  has a value

bed007, 2018-02-06:
If I am not mistaken, the Device Name is populated from the advertisement packets. So is it possible that your BLE device does not provide a name ?

Are you using a commercial device or are you developping your own ?

PawelGasiorowski, 2018-02-07:
Hi,

Update from my side. As deep as I dug into this, the device name might be passed through ADV record data, however it is byte limited. In a result, most of devices that I tested didn't have it. IOS has built in mechanism to handle this - it has some system cache, not possible to clear, that pairs device IDs (mac address accessible from system layer or simple GUID) with the name that IOS gets automatically from Device info characteristic just after first connection. Result is that even if the ADV record does not contain device name, IOS knows what it is, looking at the cache. Of course if no connection has been established IOS can't use the magic to guess the name, but it is good to know this cache exists when testing any solution :)

Android doesn't have a cache, so the name is always empty.
I ended up coding the cache inside the application to cover Android.

cheers,
Pawel

Malik-Jehangir, 2018-02-24:
@PawelGasiorowski  How did u do it? Can u brief me cuz Im new to this and my application opens on android, although my bluetooth is on but I still unable to see my polar H7 heart rate monitor?

Malik-Jehangir, 2018-02-24:
Hi all

Ive got it working, it is showing all the characteristics of the heart rate monitor except the heart rate.. Any idea why?

bapgobears, 2020-04-25:
> @DumDumin
> 
> From our readme : )
> Add these permissions to AndroidManifest.xml. For Marshmallow, please follow [Requesting Runtime Permissions in Android Marshmallow](https://blog.xamarin.com/requesting-runtime-permissions-in-android-marshmallow/)
> 
> ```
> <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
> <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
> <uses-permission android:name="android.permission.BLUETOOTH" />
> <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
> ```
> 
> Android >=6 requires location permission in order to perform BLE discovery (this is because of iBeacons) and needs the user to grant this permission. Just adding those lines to your manifest without permission request **will not be enough.** You still need to go to Apps>YourApp>Permissions and enable Location. Follow the link to see how this is done. In the future this might become part of the plugin, but for now this is up to the developer.

Hi Guys,

I think we could save a lot of headache for people getting started if on the FRONT PAGE of the repository after describing the MANIFEST changes for Android, add the sentence, "Android >=6 REQUIRES you to enable Location permissions on your device for your app.  Go to Settings/Apps/YourApp/Permissions and enable Location".  If you don't do this, the scan will not find any devices.

Thanks!
Brett

Kikanye, 2021-05-19:
> Hi,
>
> Update from my side. As deep as I dug into this, the device name might be passed through ADV record data, however it is byte limited. In a result, most of devices that I tested didn't have it. [...]

How did you do this?

PawelGasiorowski, 2021-05-20:
Use device mac address as a key (available only in android) and name from device Info characteristic as a value. Alternatively you can check if Android doesn't do it automatically from deviceInfo.

janusw, 2021-05-20:
I think this issue can be closed. The README explains the runtime permissions requirements for Android 6+. For any further problems discussed here, please open a new issue if necessary.

janseris, 2023-03-14:
Not only you need permission but also turned on GPS to find any device.